blueman-project/blueman #782: Any plans on Python v3.6?

p-boiko, 2017-10-18:
blueman: 2.1alpha1
BlueZ: 5.47-1
Distribution: Debian (Sid)
Desktop environment: Openbox

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi!

Debian moved their Python3 packages to v3.6. Most of the packages, dependent on the Python3 didn't resist, but the Blueman depends on v3.5 and it withholds from total system upgrade. Is this dependency dictated by Blueman's code or I should question a Debian package maintainer for extending Python version to v3.6?

Thank you.

parkerlreed, 2017-10-18:
You should just be able to rebuild it. There's nothing tying it to 3.5 specifically. Arch has been on 3.6 and I've installed from git plenty of times and it works fine.

infirit, 2017-10-18:
I have been using master with 3.6 and it is fine. It most likely just needs a rebuild as @parkerlreed said.

cschramm, 2017-10-25:
That's a pure packaging thing, yes. The version in sid [has been rebuilt for python3.6 as 2.0.5-1+b1](https://buildd.debian.org/status/package.php?p=blueman) as part of [the python3.6-default transition](https://release.debian.org/transitions/html/python3.6-default.html). The version you have, 2.1~alpha1-1, is from experimental and thus not affected by the migration.

closed 2017-10-25 by cschramm