blueman-project/blueman #785: Can't compile with Python 3

phillipviana, 2017-10-27:
blueman: latest from GitHub master branch as of Oct 26
BlueZ: 5.44-4.el7_4
Distribution: RHEL 7
Desktop environment: MATE

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

I have both Python 2.7 and 3.4 installed. 

I am able to `./configure && make && make install` successfully, but when I try to run `blueman-applet` I get the following error:

```
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 15, in <module>
    from blueman.Functions import create_logger, create_parser, set_proc_title
  File "/usr/lib/python3.4/site-packages/blueman/Functions.py", line 31, in <module>
    import gi
  File "/usr/lib/python3.4/site-packages/gi/__init__.py", line 39
    print url
            ^
SyntaxError: Missing parentheses in call to 'print'
```

I believe this happens because the applet is running with Python 3.4 and the code is for Python 2.
I tried changing the first line of the program from `#!/usr/bin/env python3` to `#!/usr/bin/env python2` and then I get:

```Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 15, in <module>
    from blueman.Functions import create_logger, create_parser, set_proc_title
ImportError: No module named blueman.Functions```

I saw in the changelog that Python 2.7 is no longer supported - so sounds like Python 2 is out of support but the code still contains Python 2 syntax.

What am I missing here?

Thank you in advance.

cschramm, 2017-10-28:
The line you're referring to is part of `/usr/lib/python3.4/site-packages/gi/__init__.py`, so it's outside of blueman. Seems like you have a `gi` package in your `/usr/lib/python3.4/site-packages` that is not Python 3 compatible.

infirit, 2017-10-28:
Epel does not appear to provide pygobject version for python3.4 and without blueman from master will not work. Best to stick with [blueman 2.0.X](https://github.com/blueman-project/blueman/tree/2-0-stable). Is there a specific reason you are using master?

phillipviana, 2017-10-28:
Thanks @cschramm - that makes sense. So I think an external dependency is causing blueman to fail. @infirit thank you, I'm compiling 2.0.x now.

phillipviana, 2017-11-07:
@cschramm @infirit I compiled blueman 2.0.x and I'm facing the same problem, because blueman is being run with Python 3 and I can't get `gi` for Python 3 :-( is 2.0.x supposed to work with Python 2? Is there a way to configure that?

cschramm, 2017-11-08:
Yes, 2.0 is supposed to work with Python 2.7. You can set the python interpreter with the `PYTHON` variable, e.g. `PYTHON=/usr/bin/python2.7 ./configure`. `configure` will tell you which version it's using when it's done.

phillipviana, 2017-11-08:
I tried doing what you said and got this:

```checking whether /usr/bin/python2.7 version is >= 3.2... no
configure: error: Python interpreter is too old
```
I cloned blueman 2.0 from here: https://github.com/blueman-project/blueman/tree/2-0-stable

cschramm, 2017-11-09:
You have to `git checkout 2-0-stable`.

phillipviana, 2017-11-09:
D'oh! Of course.

That did the trick. Thanks!

closed 2017-11-09 by phillipviana