blueman-project/blueman #786: Blueman removed from EPEL

rothgar, 2017-11-03:
Distribution: CentOS 7

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

It appears blueman was removed from the EPEL 7 repositiory. What is the recommended way to install blueman on CentOS and RHEL distributions?

cschramm, 2017-11-03:
Unfortunately https://apps.fedoraproject.org/packages/blueman/overview/ looks like you're right. It's not in EPEL. I cannot find a prove though that it's been removed and why.

Did you try just downloading and installing one of the Fedora RPM files?

leigh123linux, 2017-11-15:
FTR EPEL repo never had blueman

https://koji.fedoraproject.org/koji/packageinfo?packageID=8707

I imported it but never built it

https://src.fedoraproject.org/cgit/rpms/blueman.git/log/?h=epel7

cschramm, 2017-11-15:
Doesn't https://admin.fedoraproject.org/pkgdb/package/rpms/blueman/ say that it is / has been in "Fedora EPEL 7"? 🤔

leigh123linux, 2017-11-15:
No, it's means I applied for a epel7 branch for git and koji

cschramm, 2017-11-15:
I see. Thanks for clarifying.

closed 2017-11-21 by cschramm