blueman-project/blueman #787: Sony wireless keyboard, Model no. CECH-ZKB1GB cannot connect

a34fm, 2017-11-04:
blueman: 1.23-git201403102151-1ubuntu1
BlueZ: 4.101-0ubuntu13.3
Distribution: Kernel 4.4.0-70-generic
Desktop environment: Linux Mint 17.3 Cinnamon 64-bit

My keyboard is recognized but failed to connect.

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

cschramm, 2017-11-28:
Please provide the debug log from blueman-applet.

a34fm, 2017-11-28:
Excuse me but I don't know where to find the debug lob.

cschramm, 2017-11-28:
See the debugging section on the troubleshooting page you read earlier.

closed 2017-12-13 by cschramm