blueman-project/blueman #790: Not autostarting obexd broke file transfers in non systemd distros

infirit, 2017-11-14:
I don't use obex much therefor did not notice until recently that 161b8a47875ef642f2ed782781f25a8f1ed656cf caused obexd not to be started at all for me. Gentoo does patch the obexd service file to have a proper exec entry.

Not sure what the best solution is, reverting will create an error message for systemd users that have not enabled the obexd user service. 

I am thinking to parse the error when we fail to activate obexd. But other suggestions are very welcome.

rworkman, 2017-11-23:
We patch that as well: https://mirrors.slackware.com/slackware/slackware-14.2/source/n/bluez/bluez-5.30-obexd_without_systemd-1.patch.gz
and as you see, the patch came from Arch, which suggests that it's fine there. 
IMHO, this is something that the distro should have handled, so the appropriate action for blueman would be to expect it to work.

infirit, 2017-11-23:
Thanks for the feedback @rworkman. I'll probably revert it, then catch (perhaps some parsing) the error so we can provide useful feedback to the user.

infirit, 2017-12-02:
I took different approach that does not require exception parsing. See https://github.com/infirit/blueman/commit/405f32a5f4e0b337d1f21b48612e9c73e6ad3ec1 the approach I took for the TransferService plugin. Next up sending files.

infirit, 2017-12-06:
With some additional changes this should be solved now, closing. If anyone runs into issue let me know.

closed 2017-12-06 by infirit