blueman-project/blueman #791: Reason for failure of DBus calls.

paratsa, 2017-11-15:
Hi All,

I am working on the car infotainment system.
The details of the system is given below.

Platform  : LINUX
IPC          :  Dbus
BT Stack  : Bluez
Module   : AddressBook 

When ever i am selecting the Contacts or the Call logs the below mentioned Dbus calls are invoked and became success.
dbus_message_new_method_call(OBEX_BUS_NAME, File_TransferPath,"org.freedesktop.DBus.Properties","Get");

dbus_connection_send_with_reply_and_block(session_bus, msg, -1, &err);

In some cases the above shown Dbus calls are failing and giving the below errors.

Error Message:**Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist**
Error Name:**org.freedesktop.DBus.Error.UnknownObject**

I have no clues why this Dbus call had failed.

Please help me in this regard.

Note : Both the Dbus calls are invoked to get the data from Bluez.

Thanks and regards,
Parthasarathi S

cschramm, 2017-11-28:
You probably found your way to the blueman project with this as those error messages are all over blueman's logs as well. I'm not 100 % sure what's happening there, but I think the issue is that the DBus objects are just not available when making that call, so it's basically just bad timing.

paratsa, 2017-11-28:
Thank you very much cschramm. How do I resolve this bad timing and I would like to know why the DBus objects are destroyed. Please suggest me in this regard.

cschramm, 2017-11-28:
Well, I don't know your code, but obviously the object path in `File_TransferPath` isn't valid anymore when you issue the method call.

closed 2017-11-28 by cschramm