blueman-project/blueman #793: sony mdr-1000x doesnt connect

shashank-gautam, 2017-11-21:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

cschramm, 2017-11-21:
Sooo, please consult the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting).

shashank-gautam, 2017-11-21:
I have already checked the trouble shooting page .. The device is not
displayed in the search .. Can you please help?

Regards,
Shashank

cschramm, 2017-11-21:
You describe two different issues in the title ("does not connect") and your last comment ("is not displayed"). I'll assume the latter is the case. You forgot to add you system information. I'll assume BlueZ 5.

Open `bluetoothctl` and type the command `scan on`. If the device does not show up, it's probably not discoverable. Consult the manual of the device.

shashank-gautam, 2017-11-21:
Thanks for the speedy resolution of my problem.

Regards,
Shashank

closed 2017-11-21 by cschramm