blueman-project/blueman #803: Bluetooth speaker (client) disconnected after reboot

hannesrapp, 2017-12-04:
Hello,

After a system reboot, my bluetooth speaker is always disconnected. In blueman-manager devices menu, the speaker is all of the following: VISIBLE, PAIRED and TRUSTED.  In my startup apps, blueman-applet is included.

To connect a device after reboot, I just open device menu in blueman manager and follow the "setup" procedure which now starts at step 4 "Connect" phase. This works without any issues. I want to avoid having to follow this procedure every time I do a system reboot. If there's anything else that must be done for a persistent connection, please let me know. If theres a command line alternative to this procedure, that would be acceptable as a workaround (can include in startup).

Regards,
Hannes

+++

blueman: 2.0.4 Blueman-manager
BlueZ: none
Distribution: Xubuntu 17.10
Desktop environment: XFCE4

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

cschramm, 2017-12-05:
A right click on the device should give you a "connect" entry you can use.

A CLI automation would be something like `echo 'connect YO:UR:DE:VI:CE:ID' | bluetoothctl`.

There is a ticket for an auto-connect feature in blueman but it's currently not in the pipeline: #239 (also see the discussion in https://github.com/blueman-project/blueman/issues/775#issuecomment-333956546).

hannesrapp, 2017-12-05:
thx @cschramm

closed 2017-12-05 by hannesrapp