blueman-project/blueman #804: ctrl-c fixes landed in pygobject

infirit, 2017-12-06:
The following commits indicate that we may be able to get rid of the workaround in the app/* scripts.

https://github.com/GNOME/pygobject/commit/a321f6e9d8f5b8e779892eab4ce759b60ff98e39
https://github.com/GNOME/pygobject/commit/58f677bfaa0f117465a9e2146c5d83768b5a76ac

Not in a released version yet but we should keep follow up at some point when it landed in most distros.

infirit, 2018-01-20:
The version this landed in is 3.27.0

cschramm, 2018-03-26:
Debian just got 3.28.1 in unstable.

Gentoo, Arch Linux, Fedora Rawhide, and openSUSE Tumbleweed do have 3.28.1 as well.

So packages are available and just have to propagate into stable releases in Debian, Fedora, openSUSE, and derivatives.

infirit, 2018-03-27:
Still a problem unfortunately even after those commits.

infirit, 2018-05-01:
>the fixes do not actually help us

From my tests it doesn't no.

>just close this issue

Yup, closing.

closed 2018-05-01 by infirit