blueman-project/blueman #807: cannot auto connect to iphone as audio source

TrackLiao, 2017-12-10:
Hi,
   I'm currently doing a small project that use my iPhone to unlock my laptop, however when my phone out of range or restart bluetooth on my phone, blueman will not automatically connect my phone, which make my unlock failed. 
   I need to get in to GUI and there are two connection choice: Connect to Audio source and Connect to Network Access Point. And I need to select Audio source everytime
   How can blueman automatically connect my phone as audio source? is there a profile that can define default method for selected devices?

TrackLiao, 2017-12-10:
Well I solve this problem by reading one of the closed Issues suggetion   `echo 'connect UUID' | bluetoothctl` and add it to my script, it works fine but I think this kind of default selection should add to blueman

infirit, 2017-12-10:
There is a issue open for autoconnect, see #239

closed 2017-12-10 by infirit