blueman-project/blueman #808: Bluetooth Devices connected but no sound coming from devices

ssunkev, 2017-12-12:
blueman:
BlueZ:
Distribution:
Desktop environment:
Hi All, 
I have freshly Downloaded UBUNTU MAte, to try to solve all of my Computing simple issues.
However I am not super litterate and don't know how to write commands that do things that I used to do by just clicking install or run. 
Anyway, I have connected a BOSE Soundlink mini II and a Bluetooth headset, all seems to be fine in the pairing but no sound comes from them, although they also show that they are connected to my Laptop.
I have tried a few things from the options given , but could not get them to work.

Can someone please help?

Many thanks
* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

infirit, 2017-12-21:
You do not specify which version of ubuntu and blueman you use. But most problem folks have with audio are covered by https://wiki.debian.org/BluetoothUser/a2dp or https://wiki.archlinux.org/index.php/Bluetooth_headset (both taken from the TroubleShooting page!).

closed 2018-01-07 by cschramm

ssunkev, 2018-01-08:
Hi

Thanks for your reply.

I am having trouble understanding the information contained in the link below.

It seems to me that the text in the grey boxes refers to commands, but if I use them in the Terminal, it doesn't work, and I do not know what else to do with it.

However my OS is Ubuntu MATE 16.04, and blueman applet 2.0.4

It pairs succesfully but does not connect.

Last message : Connection Failed: blueman.bluez.errors.DBusFailedError: Host is down

I guess I have a few things to learn to be able to use and tweak my ubuntu properly to get it to work.

Thanks again for your help

regards

Kevin LOMBA

cschramm, 2018-01-14:
Try if `sudo apt-get install pulseaudio-module-bluetooth` improves your situation (pay attention to if it actually installs that package or just says it's already installed).

ssunkev, 2018-01-21:
Hi and thanks again for your help.

I have tried the command sent and no difference to my issue, the device connects to the laptop, but no sound is transmitted.

Here is the message that I get :

Reading package lists... Done
Building dependency tree
Reading state information... Done
pulseaudio-module-bluetooth is already the newest version (1:8.0-0ubuntu3.7).
The following package was automatically installed and is no longer required:
  libllvm4.0
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

From there, I Have restarted my computer, still no improvment. Don't know what to do.
What I can say is that I was using UBERMIX previously, which is still installed on my laptop, and on that OS Bluetooth worked no problem, without particular setup. However it crashed a bit too often for my taste.

I don't know if this information helps, and I hope that you can help me further resolving this issue.

Thanks in advance.

This is a great community