blueman-project/blueman #809: Manjaro bluetooth. I can send from my laptop to my mobile but not from mobile to laptop

take5or1, 2017-12-13:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

cschramm, 2017-12-13:
https://github.com/blueman-project/blueman/wiki/Troubleshooting

infirit, 2017-12-21:
Go to the link @cschramm provided and post the output of blueman applet run from a terminal. It will show if we actually see the transfer coming in or not and possibly errors.

closed 2018-01-07 by cschramm