blueman-project/blueman #817: recover removed device

ag3ntsim0, 2017-12-31:
hi,
i have accidentally removed a device from gui interface ! 
how can i recover the device to be shown in the scan list?

infirit, 2018-01-01:
The device is removed permanently so not possible AFAIK.

cschramm, 2018-01-01:
I don't think there's a (convenient) way to manually add a device to BlueZ either. You can only get new entries by discovering devices.

closed 2018-01-01 by cschramm

bigfoot31, 2019-04-24:
This is shocking. So a device accidentally removed, will never appear on the scan menu?

cschramm, 2019-04-24:
@bigfoot31 No, of course you can re-discover the device.

l0b0, 2019-05-07:
After a reboot and starting Blueman the device is still gone. I've been using it for about a year almost daily, and now Blueman has somehow removed it permanently even from `bluetoothctl`. Discovering does *not* help.

rjurado01, 2019-08-02:
Usually this is problem of your device. You should reset pair info in your device and set in pair mode again. This was my case with a Sony Headphones.

tamirOK, 2019-09-06:
Hi!
I've faced the same problem. I have removed my headphones from gui and search does not detect it. 

I am using Blueman 2.0.5, Linux Mint 19.01 with Cinnamon

taguhiM, 2021-03-04:
Have you found out any convenient way to do this?

matiasvlevi, 2022-06-03:
I got my device back by deleting the packages `bluez bluez-utils pulseaudio-bluetooth` and then restarting the service with `systemctl restart bluetooth`. This is mildly infuriating, bluetoothctl should warn you before you try removing any devices, or at least have a more precise description in the help menu, the command can easily be confused with "disconnecting" or "unpairing" a device. This issue should remain open in my opinion

cschramm, 2022-06-03:
@matiasvlevi: Removing a device does not mean that it cannot get re-discovered. Something's not working as expected if that's the case.

If reinstalling BlueZ packages is a solution, it seems like an issue with BlueZ' persisted state that probably gets removed by your package manager. See the [storage's documentation](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/settings-storage.txt) for reference. As you can see there are two device-specific locations: In a cache directory and one level up. When I remove a device and then gracefully stop bluetoothd, the latter one gets removed and only the cache stays in place. It does not contain anything too interesting, only e.g. a device name, possibly service data and things like that, so I don't see how it could prevent discovery, but if that's the only data left for the device I guess removing it does the trick. :shrug:

ugmurthy, 2022-07-29:
@all,
I faced a similar problem. Here are the details as to how it was resolved.

How did the issue occur?
I used `bluetoothctl` to remove a custom BLE device. After that it was impossible to discover this device via `blutoothctl` and my own python program that used `dbus` module. 

I tried the following - none of which worked.
- restarted `bluetooth` daemon and then `scan on` using `bluetoothctl`
- tried pairing the device using `bouetoothctl`
- restarted my BLE device

Finally I tried the following which worked:
- using details in [storage's documentation](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/settings-storage.txt) copied an existing device folder with a name corresponding to my missing device folder 
- deleted `attributes` file
- modified file `info` to match my device info just 4 line relating to:
     Name, AddressType, Trusted, Blocked
- used bluetoothctl to connect to device 

and it worked!

psathyrella, 2022-11-21:
This is similar to what folks above have suggested, but maybe fewer steps. I just removed/moved the device's dir in `/var/lib/bluetooth`, then restarted `service bluetooth`.

3lilspins, 2023-04-22:
@psathyrella , perfect, thank you

WhiteBlackGoose, 2023-04-22:
Got a notification from this and saw my old comment made a bit out of anger here. Sorry. This is great software. I was just very frustrated when I was writing the comment :sweat_smile:

3lilspins, 2023-04-22:
understood. tbh I also was very annoyed by the issue!

blueglyph, 2024-09-24:
I had the same problem with a keyboard / mouse devices. After connecting, trusting, and pairing the devices, they were visible in blueman but not responding to keypresses or mouse movements / clicks.

I removed the mouse to see if reconnecting would help. From that moment on, I never saw it again in the list (despite rebooting, scanning, changing the mouse's bluetooth channel, etc).

Also, the Search button became greyed-out and unusable, though I could still access it from the menu Adaptor / Search. (?)

I discovered that after rebooting, the keyboard started to work. The Search button was usable again, too.

After trying a few other things, I finally

- stopped the bluetooth service
- uninstalled bluez bluez-libs bluez-utils blueman pulseaudio-bluetooth manjaro-pulse
- deleted `/var/lib/bluetooth/<adapter ID>` (fortunately, there's only one, which made it easier)
- reinstalled them
- started the bluetooth service
- scanned the devices

... and it worked at last. Maybe not everything was needed.