blueman-project/blueman #818: Add device error for broadcom

akshay4497mane, 2018-01-01:
Hi,
I am using newest version of Lubuntu 16.04 which has blueman-manager 2.0.4

I came across this error while connecting my Plantronics Bluetooth Headset to my Lenovo G550 Laptop(broadcom chip)

Heres the error i got:
"Add a bluetooth device Error:     Connection Failed: blueman.bluez.errors.DBusFailedError: Protocol Not available"

I could resolve this temporarily by following these steps, 
I wonder if this could this be included in the original source code itself?

akshay@ubuntu:~$ sudo apt-get install pulseaudio-module-bluetooth
akshay@ubuntu:~$ pactl load-module module-bluetooth-discover
If it works, you can consider adding the second command to your startup settings, so that you don't have to run it again after every reboot.
How?
Ans:  Start MENU > Preferences > Default Applications for LXSession > Autostart > Click on  + Add(pactl load-module module-bluetooth-discover )

https://askubuntu.com/questions/801404/bluetooth-connection-failed-blueman-bluez-errors-dbusfailederror-protocol-no
https://zach-adams.com/2014/07/bluetooth-audio-sink-stream-setup-failed/

blueman:2.0.4
BlueZ:
Distribution:Lubuntu 16.04
Desktop environment:  LXDE

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

infirit, 2018-01-01:
Firstly, this is mostly a distribution problem and we have an entry for [A2DP](https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect) in troubleshooting.

After installing ``pulseaudio-module-bluetooth`` normally pulseaudio should load it automatically and there is no need to run ``pactl`` on every new session. The configuration for this is in ``/etc/pulse/default.pa`` Check for discover and bluetooth, it should like similar to the below.

```
### Automatically load driver modules for Bluetooth hardware
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.endif

.ifexists module-bluetooth-discover.so
load-module module-bluetooth-discover
.endif
```

closed 2018-01-15 by cschramm

parkerlreed, 2018-01-16:
@infirit Hah, I had forgotten about writing that little GDM tidbit on the wiki. Nice :D