blueman-project/blueman #819: ubuntu 17. JBL GO speaker not listing

pinto-xavier-dev, 2018-01-07:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

infirit, 2018-01-12:
I am not entirely sure what your problem is. Most audio related problem are handled in https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect so you may want to have a look through that.

If that did not help please described the problem and what steps you took, what you expected and what actually happens.

infirit, 2018-01-20:
Closing for now, feel free to comment and we can reopen if there still is a problem.

closed 2018-01-20 by infirit