blueman-project/blueman #820: Regarding New Message Notification

paratsa, 2018-01-08:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

paratsa, 2018-01-08:
REGARDING NEW MESSAGE NOTIFICATION :

Hi All,

I am working on the below mentioned environment.

Operating System : Linux
Interprocess Communication : Dbus
Bluetooth Stack : Bluez

I am working on the Message Service. I am able to fetch and populate and Messages in HMI Via MAP profile. The problem what i am facing is , once the phone (pairing and )connection is over, if the mobile connected to the board gets the new messages run time , there is no notification triggered from Bluez to the Message Service hence run time new messages are not populated in HMI. 

I request you to guide/suggest me how to retrieve the new messages. Which is the interface i suppose to listen to or which signal will be triggered by Bluez for the new messages.

Thanks and regards,
Parthasarathi S    
+91 9731580005

infirit, 2018-01-12:
Hi @paratsa, I think you may have us confused with the [BlueZ project](http://www.bluez.org/). And if I understood you correctly they have documentation on MAP dbus interface here: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/obex-api.txt

closed 2018-03-27 by infirit