blueman-project/blueman #821: Not pairing: Authentication Failed

pprw, 2018-01-08:
blueman: 2.0.5
BlueZ:  5.47-1+b1 (debian package)
Distribution: Debian Sid
Desktop environment: Xfce

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I cannot pair a device anymore (the same device works fine until recently). 

I have an error regarding the authentication:

```
Destroy (/usr/lib/python3/dist-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_88_C6_26_AD_C2_6A 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 607, in msg_reply_handler
    *message.get_args_list()))
  File "/usr/lib/python3/dist-packages/blueman/bluez/Device.py", line 33, in err
    raise exception
blueman.bluez.errors.DBusAuthenticationFailedError: Authentication Failed
```

complete output of blueman-applet: https://paste.aperture-sc.net/?9a8fd83de8d2e9d3#7p4jpFn0lchP/nWfrYAoT5Guz76n+0WBiAGMHWpC0iE=

inukaze, 2018-01-11:
blueman: 2.0.4
BlueZ: 5.47
Distribution: Slackware64 14.2
Desktop: XFCE

Hi there i am trying to pair my smarthphone (DG-310) with a bluetooh adapter (with lsbusb don't show any name or something similar)

```
# lsusb -v
Bus 002 Device 007: ID 1caa:0001  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          224 Wireless
  bDeviceSubClass         1 Radio Frequency
  bDeviceProtocol         1 Bluetooth
  bMaxPacketSize0        64
  idVendor           0x1caa 
  idProduct          0x0001 
  bcdDevice           19.15
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          193
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
[... 208 lines cut ...]
  (Bus Powered)
```

i try to use hciconfig : 
```
hciconfig -a hci0
hci0:	Type: Primary  Bus: USB
	BD Address: AA:BB:CC:56:34:12  ACL MTU: 339:6  SCO MTU: 180:1
	UP RUNNING PSCAN ISCAN 
	RX bytes:3112 acl:0 sco:0 events:277 errors:0
	TX bytes:1332 acl:0 sco:0 commands:159 errors:0
	Features: 0xef 0x3e 0x09 0xf0 0x0b 0x08 0x00 0x80
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'BlueZ 5.47'
	Class: 0x100104
	Service Classes: Object Transfer
	Device Class: Computer, Desktop workstation
	HCI Version: 1.2 (0x2)  Revision: 0x2
	LMP Version: 1.2 (0x2)  Subversion: 0x2
	Manufacturer: Accel Semiconductor Ltd. (74)
```

with blueman-manager i select device and i clic pair, and nothing happened, from the smartphone i try to pair with "Bluez 5.47" i put 0000 or 1234 and failed. which are the pin to connect it ???

infirit, 2018-01-12:
@pprw the paste is gone. Can you paste it again maybe to a github gist which does not expire. Also make sure to post the log from till after you tried pairing and it failed.

But a question, do you see a notification with the pairing information? if not this may indicate no notification daemon is installed or has problem showing the notification.

We do have a fallback dialog window which you can force by setting a gsettings key. From a terminal
```
gsettings set org.blueman.general notification-daemon false
```

pprw, 2018-01-13:
@infirit 

I just tried another usb port for my bluetooth adapter and it works !

Sorry for the paste (the default expiration time is 24h). Here is one with no expiration date I uploaded before I tested a new usb port. I put it for information but I don't have the problem anymore.

https://paste.aperture-sc.net/?c5928144a6d26358#Ef7/H/EbR+8GqpZqqddPDi7C9d41NXde5xHmYsCE7f0=

infirit, 2018-01-20:
>I just tried another usb port for my bluetooth adapter and it works !

This may be because there are too many usb devices on a hub. But good to see it got fixed :+1:

closed 2018-01-20 by infirit