blueman-project/blueman #826: Connection Failed: blueman.bluez.errors.DBusFailedError: Input/output error...

emiliskvg, 2018-01-24:
blueman: 2.0.5-1
BlueZ: 5.48-1
Distribution: Arch Linux (up-to-date)
Desktop environment: XFCE

I have this weird issue with my Bluetooth.
When I first boot up, I can connect to my Bluetooth speaker and play audio trough it with no issue. But if I disconnect the Bluetooth speaker and try to connect, it doesn't connect at all. (a reboot solves this)

When trying to connect I get the "Connection Failed: blueman.bluez.errors.DBusFailedError: Input/output error" generic error message. I tried looking at the [logs](https://pastebin.com/TB6XYbg3) but, they didn't get me far.

I'm using a ThinkPad X230
The Bluetooth adapter is:
BCM20702A1 (001.002.014)
ID: 0a5c:21e6
Bluetooth version: 4.0
FRU: 60Y3305 Liteon
firmware file: brcm/BCM20702A1-0a5c-21e6.hcd
(I tried using a few different builds of this firmware file and it's still the same, even extracted the hex firmware from lenovo website and then converted with hex2hcd)

[Output from dmesg when powering on Bluetooth(turning it ON)](https://pastebin.com/BdRZ0m1F)

[X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

[Pastebin of $ blueman-applet --loglevel debug](https://pastebin.com/TB6XYbg3)

cschramm, 2018-01-25:
`Input/output error` hints to all sorts of issues with specific hardware / firmware / drivers. Some examples are in #650, #413, #240. Disabling the adapter or repairing the remote device could fix the issue temporarily.

You might get more useful information about what goes wrong by stopping bluetoothd and starting it with the `-d` flag to send debugging information to syslog.

emiliskvg, 2018-02-04:
Followup: I tried debugging the BCM20702A1 adapter for weeks now, nothing helped. So I went out and got a fairy cheap "LogiLink BT0015 USB bluetooth V4.0 EDR Class1 Micro, CSR Chip", which gets recognised fine. I can search bluetooth devices with it and add/pair them too, but it has the same issues....

There are only two errors that I encounter:
>blueman.bluez.errors.DBusFailedError: Input/output error   

and    

>blueman.bluez.errors.DBusInProgressError: In Progress

But with this adapter, after getting like ~2-4 errors (trying to connect as Audio sink) it connects eventually :D So I'm fairly sure that there is some issue with my packages or configs.

closed 2018-02-05 by cschramm