blueman-project/blueman #829: Main window does not open in a certain scenario

irgendsontyp, 2018-01-27:
blueman: 2.0.5
BlueZ: 5.48
Distribution: Arch Linux
Desktop environment: XFCE 4

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

After boot, bluetooth.service is disabled and not running. I start it via `$ sudo systemctl start bluetooth`. Right after that, I run `$ blueman-manager --loglevel debug` in a terminal. A dialog appears, saying that bluetooth is not enabled. It offers a button "Enable". The main dialog is not visible anymore at this point. After having clicked "Enable", the dialog disappears. Now, when I click on the notification area icon of bluetooth-applet, the main window does not appear. The cursor just turns into the "arrow + clock" icon but the main window never appears. When I kill blueman-applet and blueman-manager and restart blueman-manager after that, everything works fine: The main window appears and bluetooth works like a charm.

The output from starting blueman-manager until having clicked the applet icon can be found [here](https://gist.github.com/irgendsontyp/5db970d712e4ce142abe0724918ab53c).

infirit, 2018-01-28:
Yes 2.0.x does not handle this very well. We are working to make the next version, 2.1, handle this properly. If you are willing to try what will become 2.1 modify https://aur.archlinux.org/packages/blueman-git/ and update the git hash to the most recent one.

infirit, 2018-01-28:
That aur package it is very out of date and will need work to get working.

irgendsontyp, 2018-01-30:
When will 2.1 be released?

infirit, 2018-01-31:
That will probably take a bit more time. But current master is quite usable and you can try out the build I just did. Get it on https://www.dropbox.com/s/40zgxszlu082p06/blueman-2.1.203ecbf-1-x86_64.pkg.tar.xz?dl=0

irgendsontyp, 2018-02-03:
I guess the fix for 2.0.x is not just about showing blueman's main window? Based on whether the answer is yes or no, I'd might want to take a look into the code and try to fix this.

infirit, 2018-02-03:
Its not. The main problem is with blueman-applet not handling the way systemd does dbus things. We worked on this in 2.1/master and I put the finishing touches on it last week testing on arch. The result is above build (I can post the PKGBUILD and files if you want).

The way the code works is that it checks if bluetooth is running properly and then quits. The blueman-manager code in master technically does not need to exit anymore but we need to discuss and decide if this is something we want. This is why I put on the enhancement label.

The code in master is actually much cleaner and works much better than what was in 2.0. And I would love some testers, outside the two of us, who can run it and report how things are working for them.

cippaciong, 2018-10-27:
@infirit I had the same problem with version 2.0.6 in Arch. I installed `blueman-git` now and it's working fine :+1:

closed 2019-04-27 by cschramm