blueman-project/blueman #830: Speaker can pair but not connect Debian 8

TheZenKitteh, 2018-01-28:
blueman: 1.99.alpha1
BlueZ: 5.23-2+deb8u1 ?
Distribution: Debian 8
Desktop environment: XFCE

* [x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

https://pastebin.com/raw/TsW62EKA

Basically, pairing seems to work, connecting doesn't. The error is: No such file or directory. I can't figure it out.

infirit, 2018-01-28:
This requires the pulseuadio bluetooth module and pulseaudio running, see [1] for the details on debian. Are they installed and is pulseaudio running?

[1] https://wiki.debian.org/BluetoothUser/a2dp

TheZenKitteh, 2018-01-28:
I followed the instructions at your link, pulseaudio won't start. Tried troubleshooting that, can't find any information. Why is pulseaudio needed?

infirit, 2018-01-28:
>Why is pulseaudio needed?

It does not have to be pulseaudio. BlueZ does not route the audio itself and relies on a 3rd party. That 3rd party can be anything but most commonly pulseadio is used. There is also a project that does something similar but much more low level which is [bluealsa](https://github.com/Arkq/bluez-alsa/).

As you said you are not familiar with audio under linux bluealsa is probably not the best option. Follow the following link to setup, install etc pulseaudio under debian. https://wiki.debian.org/PulseAudio.

TheZenKitteh, 2018-01-29:
I still can't connect to the speaker. I used this opportunity to upgrade to Debian 9 with Mate desktop (comes with pulseaudio and I couldn't get pulseaudio to work on Debian 8). Everything is supposedly as it should be, but now the error is: Connection failed: Blueman.Bluez.errors.DBusFailedError:Protocol not available. Again, pairing works, connection doesn't. Still trying to figure this out.

infirit, 2018-01-30:
Good you found that solution. In a similar issue that referenced [that thread](https://github.com/blueman-project/blueman/issues/818#issuecomment-354651184) I replied with some additional info. You should  not need to add pactl commands to startup. Just make sure the pulse config is setup correctly.

closed 2018-01-30 by infirit