blueman-project/blueman #836: Systemd again

infirit, 2018-02-02:
Testing in Arch we fail to autostart dbus services and crash again :crying_cat_face: . Too annoyed right now but will fix this weekend.

```
$ blueman-applet 
blueman-applet version 2.1.alpha2 starting
Stale PID, overwriting
Traceback (most recent call last):
  File "/bin/blueman-applet", line 42, in <module>
    BluemanApplet()
  File "/usr/lib/python3.6/site-packages/blueman/main/Applet.py", line 24, in __init__
    self.Manager = bluez.Manager()
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Manager.py", line 16, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Manager.py", line 38, in __init__
    self.__bus_name, '/', None, None, None)
GLib.Error: g-io-error-quark: Error calling StartServiceByName for org.bluez: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit dbus-org.bluez.service not found. (36)
```

cschramm, 2018-02-03:
Sounds like an issue with your BlueZ setup. A shot in the dark:

You have something like `/usr/share/dbus-1/system-services/org.bluez.service` and it includes `SystemdService=dbus-org.bluez.service`. You use systemd, so it picks up the job. For some reason, you do not actually have that service defined (at `/usr/share/dbus-1/system-services/org.bluez.service`).

infirit, 2018-02-03:
Yes this is what systemd does when you disable the bluetooth service. I reverted too much in 326a60b20589eb93db30f9c4ab1ed741e0c91f1d. I'll push a fix now to not autostart the objectmanager. Obex is ok.

infirit, 2018-02-03:
05eea0db0cacf888629d7d23ca62c73d663e98f2

closed 2018-02-03 by infirit