blueman-project/blueman #837: A2DP audio device PAIRED but fails to connect

puertas12, 2018-02-02:
blueman: 2.0.4
BlueZ: 5.43
Distribution: Debian 4.9.65-3+deb9u1 (2017-12-23) x86_64 GNU/Linux (Debian 9)
Desktop environment: lightdm-xsession

I read the two suggested posts in  A2DP audio device PAIRED but fails to connect:
[(1)](https://wiki.debian.org/BluetoothUser/a2dp) This happens because GDM captures A2DP sink on session start. However I do not use gnome but I checked if the problem is that the module "module-bluetooth-discover" in not loaded:
% pactl list | grep -i module-bluetooth-discover
> Nombre: module-bluetooth-discover
So it seems that it is loaded
[2](https://wiki.archlinux.org/index.php/Bluetooth_headset)
***Bluetooth headset
**Headset via Bluez5/PulseAudio
*Configuration via CLI:
1.- In a shell I executed:
% bluetoothctl
> [NEW] Controller CC:AF:78:8E:7A:5E debianDell [default]
> [NEW] Device CC:9F:7A:A5:6A:E6 Nokia 8
> [NEW] Device FC:58:FA:00:4C:43 SBR 4.1 A1
> Agent registered
2.- In the opened shell I executed:
# power on
> Changing power on succeeded
3.- and then executed:
# agent on
> Agent is already registered
4.- and the executed:
# default-agent
> Default agent request successful
5.- and executed:
# scan on
> Discovery started
> [CHG] Controller CC:AF:78:8E:7A:5E Discovering: yes
> [CHG] Device FC:58:FA:00:4C:43 RSSI: -79
> [CHG] Device FC:58:FA:00:4C:43 TxPower: 4
6.- Finally executed
# connect FC:58:FA:00:4C:43
> Attempting to connect to FC:58:FA:00:4C:43
> Failed to connect: org.bluez.Error.Failed
7.- In another shell I killed the pulseaudio process:
% pulseaudio -k
8.- I tried again:
# connect FC:58:FA:00:4C:43
> Attempting to connect to FC:58:FA:00:4C:43
> Failed to connect: org.bluez.Error.Failed
9.- I had a look at the log via one of the following commands:
% systemctl status bluetooth
> active (running)
% journalctl -n 20
"No traces of bluetoothd"

Do you know where to find the bluetoothd traces?
Does anyone suggest me any to find the root cause of the problem?

cschramm, 2018-02-03:
You can stop the bluetooth service and start bluetoothd manually with the `-d` switch to get debugging output in the syslog.

infirit, 2018-02-03:
With journalctl use ```journalctl --unit=bluetooth```

puertas12, 2018-02-06:
@cschramm I tried to stop the service:
`sudo systemctl stop bluetooth`
Then I wanted to confirmed that the service is stopped:
`sudo systemctl status bluetooth`
However the service is active (running).
The latter command displays the following error trace:

> bluetoothd[515]: Failed to obtain handles for "Service Changed" characteristic

puertas12, 2018-02-06:
@infirit Thanks for the command. These are the traces displayed:

> feb 06 20:35:20 debianDell systemd[1]: Starting Bluetooth service...
> feb 06 20:35:20 debianDell bluetoothd[515]: Bluetooth daemon 5.43
> feb 06 20:35:20 debianDell systemd[1]: Started Bluetooth service.
> feb 06 20:35:20 debianDell bluetoothd[515]: Starting SDP server
> feb 06 20:35:20 debianDell bluetoothd[515]: Excluding (cli) sap
> feb 06 20:35:20 debianDell bluetoothd[515]: Bluetooth management interface 1.14 initialized
> feb 06 20:35:20 debianDell bluetoothd[515]: Failed to obtain handles for "Service Changed" characteristic
> feb 06 20:36:11 debianDell bluetoothd[515]: Endpoint registered: sender=:1.35 path=/MediaEndpoint/A2DPSource
> feb 06 20:36:11 debianDell bluetoothd[515]: Endpoint registered: sender=:1.35 path=/MediaEndpoint/A2DPSink
> feb 06 20:47:32 debianDell bluetoothd[515]: connect error: Host is down (112)
> feb 06 20:47:42 debianDell bluetoothd[515]: Endpoint unregistered: sender=:1.35 path=/MediaEndpoint/A2DPSource
> feb 06 20:47:42 debianDell bluetoothd[515]: Endpoint unregistered: sender=:1.35 path=/MediaEndpoint/A2DPSink
> feb 06 20:47:42 debianDell bluetoothd[515]: Endpoint registered: sender=:1.52 path=/MediaEndpoint/A2DPSource
> feb 06 20:47:42 debianDell bluetoothd[515]: Endpoint registered: sender=:1.52 path=/MediaEndpoint/A2DPSink
> feb 06 20:47:58 debianDell bluetoothd[515]: connect error: Host is down (112)

Any ideas of what can be wrong?

infirit, 2018-02-06:
>Host is down (112)

This indicates the audio device can not be reached as if it is turned off. My mi speaker sometimes does this and I have to put it in pairing mode again.

puertas12, 2018-02-07:
@infirit I put it in pairing mode again:
`# pair FC:58:FA:00:4C:43`

> Attempting to pair with FC:58:FA:00:4C:43
> [CHG] Device FC:58:FA:00:4C:43 Connected: yes
> [CHG] Device FC:58:FA:00:4C:43 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
> [CHG] Device FC:58:FA:00:4C:43 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
> [CHG] Device FC:58:FA:00:4C:43 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
> [CHG] Device FC:58:FA:00:4C:43 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
> [CHG] Device FC:58:FA:00:4C:43 ServicesResolved: yes
> [CHG] Device FC:58:FA:00:4C:43 Paired: yes
> Pairing successful
> [CHG] Device FC:58:FA:00:4C:43 ServicesResolved: no
> [CHG] Device FC:58:FA:00:4C:43 Connected: no
`connect FC:58:FA:00:4C:43`

> Attempting to connect to FC:58:FA:00:4C:43
> Failed to connect: org.bluez.Error.Failed

> Te journal in verbose mode display the following outputs:
> feb 07 09:38:23 debianDell bluetoothd[510]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
> feb 07 09:38:23 debianDell bluetoothd[510]: src/device.c:device_bonding_failed() status 14
> feb 07 09:38:23 debianDell bluetoothd[510]: src/adapter.c:resume_discovery()
> feb 07 09:38:39 debianDell bluetoothd[510]: src/device.c:connect_profiles() /org/bluez/hci0/dev_FC_58_FA_00_4C_43 (all), client :1.51
> feb 07 09:38:39 debianDell bluetoothd[510]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_FC_58_FA_00_4C_43
> feb 07 09:38:39 debianDell bluetoothd[510]: profiles/audio/avdtp.c:avdtp_ref() 0x55ba0aff5460: ref=1
> feb 07 09:38:39 debianDell bluetoothd[510]: profiles/audio/avdtp.c:avdtp_ref() 0x55ba0aff5460: ref=2
> feb 07 09:38:39 debianDell bluetoothd[510]: profiles/audio/a2dp.c:setup_ref() 0x55ba0afea290: ref=1
> feb 07 09:38:39 debianDell bluetoothd[510]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_FC_58_FA_00_4C_43:
> feb 07 09:38:39 debianDell bluetoothd[510]: profiles/audio/sink.c:sink_connect() stream creation in progress
> feb 07 09:38:39 debianDell bluetoothd[510]: src/service.c:change_state() 0x55ba0aff2cd0: device FC:58:FA:00:4C:43 profile a2dp-sink sta
> feb 07 09:38:44 debianDell bluetoothd[510]: src/adapter.c:connect_failed_callback() hci0 FC:58:FA:00:4C:43 status 4
> feb 07 09:38:44 debianDell bluetoothd[510]: plugins/policy.c:conn_fail_cb() status 4
> feb 07 09:38:44 debianDell bluetoothd[510]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr FC:58:FA:00:4C:43 type 0 status 0x4
> feb 07 09:38:44 debianDell bluetoothd[510]: src/device.c:device_bonding_complete() bonding (nil) status 0x04
> feb 07 09:38:44 debianDell bluetoothd[510]: src/device.c:device_bonding_failed() status 4
> feb 07 09:38:44 debianDell bluetoothd[510]: src/adapter.c:resume_discovery()
> feb 07 09:38:44 debianDell bluetoothd[510]: connect error: Host is down (112)
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/avdtp.c:avdtp_ref() 0x55ba0aff5460: ref=3
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/avdtp.c:connection_lost() Disconnected from FC:58:FA:00:4C:43
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/a2dp.c:discover_cb() err 0x7ffda45888d0
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/avdtp.c:avdtp_unref() 0x55ba0aff5460: ref=2
> feb 07 09:38:44 debianDell bluetoothd[510]: src/service.c:change_state() 0x55ba0aff2cd0: device FC:58:FA:00:4C:43 profile a2dp-sink sta
> feb 07 09:38:44 debianDell bluetoothd[510]: src/device.c:device_profile_connected() a2dp-sink Host is down (112)
> feb 07 09:38:44 debianDell bluetoothd[510]: src/device.c:device_profile_connected() returning response to :1.51
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/a2dp.c:setup_unref() 0x55ba0afea290: ref=0
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/a2dp.c:setup_free() 0x55ba0afea290
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/avdtp.c:avdtp_unref() 0x55ba0aff5460: ref=1
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_FC_58_FA_00_4C_43:
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/a2dp.c:channel_remove() chan 0x55ba0afea090
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/avdtp.c:avdtp_unref() 0x55ba0aff5460: ref=0
> feb 07 09:38:44 debianDell bluetoothd[510]: profiles/audio/avdtp.c:avdtp_free() 0x55ba0aff5460

Do you have an idea of why I cannot connect?

puertas12, 2018-02-07:
I tried again pair and unpair and I got it connected. The audio is stream although sometime is breaking up. Thank you

closed 2018-02-07 by puertas12