blueman-project/blueman #839: Bluetooth manager is not responsive

simJSatw, 2018-02-10:
blueman:
BlueZ:
Distribution: Debian
Desktop environment: XFCE

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

When first installed bluez and blueman everything worked perfectly fine. Then I turned the computer off and when started it again everything stopped working. Manager opens up but i cannot click on anything. Every button is now greyed out. Bluetooth adapter app is not working properly either. It whows only 'close' sign and that's it. Sending a pic below. 
[image]

teja4477, 2018-02-27:
same problem in mint 18.2

cschramm, 2018-02-27:
Please provide the output as described on the Troubleshooting page you read.

closed 2018-03-19 by cschramm