blueman-project/blueman #843: Failed to change profile to headset_head_unit

sprawl17, 2018-02-19:
blueman: 1.99.alpha1
BlueZ: **blueman-manager**
Distribution: Debian
Desktop environment: Mate

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Yes, there was a way to add HSP/HFP to blueman but that no more works.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

You know the issue and the reason, it was your decision to neglect all the people relying on conference calls, just FIX IT.

cschramm, 2018-02-19:
> blueman: 1.99.alpha1

That's a rather old version. Please try 2.0.

>  BlueZ: blueman-manager

That's not a BlueZ version. The major version is very important to help you here.

> there was a way to add HSP/HFP to blueman but that no more works.

What happened? You seem to run a rather old system, so I guess you did not update blueman, BlueZ or Linux when it stopped working? What do you mean with "add HSP/HFP to blueman". Connecting to a related service?

The title sounds like you're seeing some error somewhere. Would you elaborate on that?

> You know the issue and the reason

No, sorry.

> it was your decision to neglect all the people relying on conference calls

No.

> just FIX IT.

Uh, yeah, very friendly.

sprawl17, 2018-02-19:
Sorry for the unfriendly tone. Missed a long awaited and important TelCo due to this sudden problem. As noted, it worked two weeks ago, now it did not. Evidently, I cannot run an update to solve an unexpected issue. I like blueman, it is usually working fine. And yes, the problem are the realaudio people who decided to drop HSP/HFP (these are bluetooth profiles) causing Linux to be unable to connect a bluetooth headsets in a way useful for TelCos. Had to install some old version, needed to adjust some system settings, frickle in settings, and than it actually worked :-) for a short time :-(. 

Now the blueman GUI asks somewhere for "report an issue", and so, in my immediate anger, I did. Couldn't blueman include the most essential bluetooth profiles not provided by anyone else, just as fall-back, a safe haven, just to avoid being left in the dark by others?

ad Version: 1.99.alpha1 is the latest provided in Debian stable repository for my release (8.10, kernel 3.16.0-5-amd64). I cannot afford unstable systems, I am doing my dayjob on it. Using virtual machines for testing, but if host system cannot connect, the client system can neither...

ad BlueZ: Where do I find the "major version" using the blueman GUI? If it's important I am sure I should find it somewhere in About... (Synaptic shows me: 5.23-2+deb8u1 - again, the latest applicable to me.)

The bug is probably not within blueman, it's a problem with supporting a specific bluetooth profile, but a lower layer problem always gets visible on the top only.

cschramm, 2018-02-19:
> the problem are the realaudio people who decided to drop HSP/HFP

You mean pulseaudio? What decision are you referring to again? Pulseaudio 11.1 *does* offer me a `headset_head_unit` profile for my headset.

> 1.99.alpha1 is the latest provided in Debian stable repository for my release (8.10, kernel 3.16.0-5-amd64). I cannot afford unstable systems

Debian jessie 8 is oldstable. The stable version is Debian 9 stretch. Also, there's blueman 2.0.4-1~bpo8+1 for you in jessie-backports.

I still cannot see you issue. Is there no "Headset Head Unit (HSP/HFP)" in blueman's audio profile menu? If you're getting an error message as the title suggests, I'd go for `pacmd set-card-profile 2 headset_head_unit` (use the right card index!) and see what that outputs. Also watch the syslog for any reports during the failure.

sprawl17, 2018-02-19:
Yes, pulseaudio, the latest provided by the repository 7.1-2~bpo8+1

And yes, seems I need to upgrade to Debian 9, but momentarily I have no time to do so. You know: "never change a running system" - at least not when you cannot afford not having a system at all...

However, as I get from your reply, Pulseaudio has corrected it's decission and supports headsets again. So I may be lucky after doing an upgrade to Debian 9. Thanks!

sprawl17, 2018-02-19:
I still cannot see you issue. 
Is there no "Headset Head Unit (HSP/HFP)" in blueman's audio profile menu? 
-> there is, but switching fails
If you're getting an error message as the title suggests, I'd go for pacmd set-card-profile 2 headset_head_unit (use the right card index!) and see what that outputs. Also watch the syslog for any reports during the failure.
-> will try to do so the next time, but hopefully there is no need at all next time...

closed 2018-02-19 by sprawl17

infirit, 2018-02-19:
The ofono project has support for the headset bits. Try to install the ofono package and make sure pulseaudio-module-bluetooth is also installed. I have not tried it on debian8 so it would be good to hear if you get it working.

>never change a running system

If it aint broke don't fix it :smile:

sprawl17, 2018-02-19:
@infirit: That I tried three weeks ago but the ofono in my 'outdated' repository did no include the headset bits. Found some hack in the internet that worked, but seemingly it's not a lasting solution.