blueman-project/blueman #845: Notifications don't disappear

tulsileathers, 2018-02-27:
blueman: 2.0.5-1
BlueZ: 5.48-2
Distribution: Arch
Desktop environment: i3wm

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

This is an issue with the blueman-applet. When choose recent connections and click a device to connect the popup notifiying me of the connection does not disappear it simply paints over whatever window I have. This happens for success and failure notifications. Closing blueman-applet also closes the notification. 

I'm not posting my whole logs as I'm pretty positive that this is the pertinent bit. Am I just missing a dependency?
```
(blueman-applet:1512): libnotify-WARNING **: Failed to connect to proxy
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
```

cschramm, 2018-02-27:
The warning sounds like you just do not have a notification daemon (i.e. any application that provides the name `org.freedesktop.Notifications` on the DBus session bus). When that's the case, blueman uses a built-in fallback based on simple (GTK+) dialogs. That does not include a default timeout for notifications (without actions) to disappear.

tulsileathers, 2018-03-04:
Yes that was it. Added dunst and that fixed it. Thank you closing.

closed 2018-03-04 by tulsileathers

rileyrg, 2023-12-29:
Anyideas if running sway/mako?  The notifications dont respect the mako timeout.

cschramm, 2024-01-01:
@rileyrg: The pairing request notification sends `expire_timeout` -1, indicating that the daemon should use its own. You could test that with `notify-send -t -1 Text`.