blueman-project/blueman #848: Can't get Blueman to work correctly (Trying to build and use it) (get_bluetooth_status is not a valid method of interface org.blueman.Applet)

tarcisiofischer, 2018-03-04:
blueman: (From master branch)
BlueZ: From apt-get (`bluetoothd -v` yields `5.46`)
Distribution: lubuntu 17.10 (https://lubuntu.net/)
Desktop environment:

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

So, I'm trying to update my blueman in order to check if I can get my new bluetooth adapter to work. It's a chinese dongle that claims it can connect to my bluetooth headphones, but I couldn't get it to work yet...

I tried blueman that cames with `lubuntu`, but couldn't get it to work, so I was trying to compile it from `master` branch in this repository in the hope I can get the newest version to work.

What I did was to install all dependencies using apt-get (Hope I didn't forget any - I know I've installed way more packages than the necessary... It was a try-and-run-again approach):

```
sudo apt-get install intltool
sudo apt-get install python
sudo apt-get install python3.6
sudo apt-get install python3-pydbus dbus
sudo apt-get install python3-dbus
sudo apt-get install python-dbus 
sudo apt-get install python-gobject-2-dev
sudo apt-get install python3-gi
sudo apt-get install python3-gi-dev
sudo apt-get install python-gi-dev
sudo apt-get install cython
sudo apt-get install bluez
sudo apt-get install bluetooth bluez bluez-tools rfkill
sudo apt-get install libpython3.6-dev libpython3.6
```

And then I've compiled using:
```
./autogen.sh
./configure
./make -j4
./make install
```

Finally, while trying to run it, I get some unexpected error... I'm hoping this is something easy to solve and someone can help out:

```
tarcisio@Fermat:/partition1$ blueman-assistant 

** (blueman-assistant:8787): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Traceback (most recent call last):
  File "/usr/local/bin/blueman-assistant", line 270, in <module>
    Assistant(args)
  File "/usr/local/bin/blueman-assistant", line 52, in __init__
    check_bluetooth_status(_("Bluetooth needs to be turned on for the Bluetooth assistant to work"), lambda: exit())
  File "/usr/local/lib/python3.6/site-packages/blueman/Functions.py", line 68, in check_bluetooth_status
    if not applet.get_bluetooth_status():
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 172, in __call__
    None)
GLib.Error: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: get_bluetooth_status is not a valid method of interface org.blueman.Applet
 (19)
```

I don't know if this is really a bug or I did something wrong... I found nothing on the Troubleshooting (https://github.com/blueman-project/blueman/wiki/Troubleshooting).

Further information that may be useful:

```
tarcisio@Fermat:/partition1$ bluetoothctl list
[NEW] Controller 00:11:22:98:76:54 Fermat [default]
Agent registered
[bluetooth]# exit
Agent unregistered
[DEL] Controller 00:11:22:98:76:54 Fermat [default]
tarcisio@Fermat:/partition1$ python --version
Python 3.6.3
tarcisio@Fermat:/partition1$ python -c "import dbus; print(dbus.version)"
(1, 2, 4)
```

tarcisiofischer, 2018-03-04:
More info that may be relevant:

```
tarcisio@Fermat:/partition1$ sudo rfkill list
[sudo] password for tarcisio: 
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
```

tarcisiofischer, 2018-03-04:
Update: Tried rebooting the system and initialize again, now I can't even start blueman-applet, it seems:

```
tarcisio@Fermat:/partition1$ blueman-applet 

** (blueman-applet:1515): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
blueman-applet version 2.1.alpha2 starting

** (blueman-tray:1522): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
blueman-tray version 2.1.alpha2 starting

** (blueman-tray:1525): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
blueman-tray version 2.1.alpha2 starting
There is an instance already running
blueman-applet 21.20.06 ERROR    BasePlugin:76 _load     : Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
GLib.GError: g-io-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached (24)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/BasePlugin.py", line 70, in _load
    self.on_load()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 24, in on_load
    self.load_nap_settings()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 45, in load_nap_settings
    m = Mechanism()
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 44, in __init__
    systembus=True, *args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached
blueman-applet 21.20.06 ERROR    PluginManager:161 __load_plugin: Failed to load Networking
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
GLib.GError: g-io-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached (24)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/blueman/main/PluginManager.py", line 159, in __load_plugin
    inst._load()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/AppletPlugin.py", line 47, in _load
    super(AppletPlugin, self)._load()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/BasePlugin.py", line 70, in _load
    self.on_load()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 24, in on_load
    self.load_nap_settings()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 45, in load_nap_settings
    m = Mechanism()
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 44, in __init__
    systembus=True, *args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
GLib.Error: g-io-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached (24)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... 7 lines cut ...]
    inst._load()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/AppletPlugin.py", line 47, in _load
    super(AppletPlugin, self)._load()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/BasePlugin.py", line 70, in _load
    self.on_load()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 24, in on_load
    self.load_nap_settings()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 45, in load_nap_settings
    m = Mechanism()
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 44, in __init__
    systembus=True, *args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Timeout was reached
tarcisio@Fermat:/partition1$ Traceback (most recent call last):
  File "/usr/local/bin/blueman-tray", line 40, in <module>
    BluemanTray()
  File "/usr/local/lib/python3.6/site-packages/blueman/main/Tray.py", line 23, in __init__
    indicator_name = applet.GetStatusIconImplementation()
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 172, in __call__
    None)
GLib.Error: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.52 was not provided by any .service files (2)
```

tarcisiofischer, 2018-03-04:
Forgot to run with debug level:

```
tarcisio@Fermat:/partition1$ killall blueman-applet
blueman-applet: no process found
tarcisio@Fermat:/partition1$ blueman-applet --loglevel debug

** (blueman-applet:1703): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
blueman-applet version 2.1.alpha2 starting
Stale PID, overwriting
blueman-applet 00.16.40 INFO     PluginManager:85 load_plugin: ['TransferService', 'AppIndicator', 'ExitItem', 'StandardItems', 'NetUsage', 'GameControllerWakelock', 'DBusService', 'PPPSupport', 'StatusIcon', 'PowerManager', 'KillSwitch', 'SerialManager', 'AuthAgent', 'DhcpClient', 'DiscvManager', 'RecentConns', 'NMPANSupport', 'Networking', 'ShowConnected', 'Menu', 'NMDUNSupport']
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 00.16.40 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Address': '00:11:22:98:76:54', 'Name': 'Fermat', 'Alias': 'Fermat', 'Class': 268, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001801-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d052E'}
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 00.16.40 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 00.16.40 INFO     Networking:31 load_nap_settings: Loading NAP settings

** (blueman-tray:1710): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
blueman-tray version 2.1.alpha2 starting
Stale PID, overwriting

** (blueman-tray:1713): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
blueman-tray version 2.1.alpha2 starting
There is an instance already running
blueman-applet 00.16.53 ERROR    BasePlugin:76 _load     : Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
GLib.GError: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/BasePlugin.py", line 70, in _load
    self.on_load()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 24, in on_load
    self.load_nap_settings()
  File "/usr/local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 45, in load_nap_settings
    m = Mechanism()
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 44, in __init__
    systembus=True, *args, **kwargs)
[... cut ...]
tarcisio@Fermat:/partition1$ Traceback (most recent call last):
  File "/usr/local/bin/blueman-tray", line 40, in <module>
    BluemanTray()
  File "/usr/local/lib/python3.6/site-packages/blueman/main/Tray.py", line 23, in __init__
    indicator_name = applet.GetStatusIconImplementation()
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 172, in __call__
    None)
GLib.Error: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.55 was not provided by any .service files (2)
```

tarcisiofischer, 2018-03-04:
After some debugging, I found out that the problem was happening while loading the `Networking` plugin. Since I don't plan making any `Networking` with bluetooth, I disabled this plugin by renaming the file `blueman/plugins/applet/Networking.py` to `blueman/plugins/applet/Networking.py.disabled`.

I'm now able to reach the applet gui, but I'm still unable to find any device using the "Search" tool...

[image]

[image]

Any help will be appreciated... Don't know how to proceed from here... I'm unsure if it's a configuration issue or my chinese dongle is just trash.

infirit, 2018-03-04:
Firstly, if you want to use current git master then use the build listed at https://github.com/blueman-project/blueman/wiki/Packaged-versions#debian-gnulinux. Please use this one before doing anything else :smiley:.

Second, Mechanism is a script that can run with elevated privileges. For example it will handle the killswitch which can not be manipulated by normal users. Also it does start a dhcpd server to handle so that your phone can connect to the hosts internet.

Lastly, if you want to get all build dependencies in debian you can use ``apt-get build-dep <package>``. The dependencies did  decrease between 2.0 and what will become 2.1 however so not all of them are needed.

Then, if an adapter is not working well it is usually not a problem in blueman but with the Bluetooth layer called BlueZ. Now lets see if we can figure something out here :smile: 

What I am missing is the output of blueman-applet when you actually try to search. And does it show any devices when you run ``bluetoothctl`` in a terminal? Example output would be ``[NEW] Device CC:61:E5:1A:39:B5 Moto G Play``.

tarcisiofischer, 2018-03-05:
@infirit Thanks for the comments.

1) Ok. I removed blueman completely (`sudo apt-get remove blueman` and `sudo make uninstall` in blueman's folder), added the third party apt repository as you suggested and installed the recent version of blueman from it (Version python 3.6).
Finally, I've debug-run `blueman-applet`, but again, some error occurs (https://pastebin.com/PaCfvW3P).

2) Well, okay. I thought it was only for networking stuff like what you mentioned about "to handle so that your phone can connect to the hosts internet", which I'm not even interested in... I don't know what is this `killswitch` you are talking about. My only goal here is to get my wireless headphones working :)

3) Nice. Thanks for the tip. Since I'm not compiling anymore (but installing from `apt` like you suggested), I'll ignore this for now.

Since it may be a problem in `bluez`, should I ask somewhere else? In `bluez`'s forums, perhaps? Or you guys are able to help me out?

Running `bluetoothctl` on terminal returns this:
```
tarcisio@Fermat:/partition1$ bluetoothctl
[NEW] Controller 00:11:22:98:76:54 Fermat [default]
[NEW] Device 34:BB:26:44:DF:07 XT1058
Agent registered
[bluetooth]# exit
Agent unregistered
[DEL] Controller 00:11:22:98:76:54 Fermat [default]
tarcisio@Fermat:/partition1$ 
```

Since I'm not able to run 'search' with the new installed `blueman`, there is nothing else to show here.
Any thoughts?

cschramm, 2018-03-05:
You can use `scan on` in `bluetoothctl`.

closed 2018-06-16 by cschramm