blueman-project/blueman #850: Feature Request: Start minimized

venomDeved, 2018-03-12:
blueman: v2.0.4
BlueZ: ?
Distribution: Linux mint (Ubuntu 16.04)
Desktop environment: Mate

* [Yes] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

I use blueman manager as my primary blue tooth application. I have it in the start up options when starting X. Can you add an option in preferences to start minimized in the desktop panel. 

Thanks

infirit, 2018-03-12:
I think there is a misunderstanding how the various blueman parts work together.

We have blueman-applet sitting in the tray as icon. Then there is blueman-manager that shows the devices, makes pairing and other operation possible. The two are separate from another and clicking on the tray icon (or use the tray menu) starts and stops blueman-manager.

venomDeved, 2018-03-12:
Thanks, you guys fixed that fast ;)

closed 2018-03-12 by venomDeved

ShuaibKunji, 2020-11-03:
> I think there is a misunderstanding how the various blueman parts work together.
>
> We have blueman-applet sitting in the tray as icon. Then there is blueman-manager that shows the devices, makes [...]

This was so helpful! Thank you so much, kind sir. You have a wonderful day ahead!