blueman-project/blueman #851: Connection Failed: blueman.bluez.errors.DBusFailedError: Protocol not available

Spudd5887, 2018-03-19:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

infirit, 2018-03-19:
There is not enough info the help. But the typical case is that you paired an audio device but then it fails to connect. This is handled in troubleshooting, https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect

closed 2018-04-17 by cschramm

Krimsun-Linada, 2018-05-16:
@infirit dude actually you are wrong. I am also having the exat same problem and i can promise you it i absolutely not at the fault of my headset. try again.