blueman-project/blueman #852: Blueman-applet pegging out CPU

Huckleberry582, 2018-03-21:
blueman: blueman-applet 2.0.5
BlueZ:
Distribution: UbuntuMate 18.04
Desktop environment: Mate
I have a Dell Latitude E6500 that I am experimenting with Ubuntu 18.04. At start-up I was noticing the fan was running like crazy and then noticed the processor was pegged out. I found that if I turned off the wifi switch, this went away and the CPU idled down. Turning the switch back on ran it back up. Turning off Bluetooth on the applet caused it to go back to normal. Turning it back on ran it back up.

I updated Ubuntu this morning after installing it. Updates did not solve the issue. I followed the directions for debugging and pasted the output below. I turned it off after the initial output and you can see the results. Please let me know if there is anything else I can provide.

mike@ITLT-UBUNTU6500:~$ killall blueman-applet
mike@ITLT-UBUNTU6500:~$ blueman-applet --loglevel debug
blueman-applet version 2.0.5 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['StatusIcon', 'NetUsage', 'PPPSupport', 'DBusService', 'NMPANSupport', 'KillSwitch', 'GameControllerWakelock', 'RecentConns', 'AppIndicator', 'DiscvManager', 'AuthAgent', 'ShowConnected', 'PowerManager', 'NMDUNSupport', 'SerialManager', 'StandardItems', 'Menu', 'Headset', 'DhcpClient', 'TransferService', 'ExitItem', 'Networking'] 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
Not loading PPPSupport because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
[... 201 lines cut ...]
on_adapter_removed (/usr/bin/blueman-applet:115)
Adapter removed  /org/bluez/hci0 
_________
initialize (/usr/lib/python3/dist-packages/blueman/plugins/applet/RecentConns.py:144)
rebuilding menu 
_________
on_adapter_removed (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:90)
/org/bluez/hci0 
_________
Run (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:208)
Function on_adapter_removed on DiscvManager Failed 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/PluginManager.py", line 212, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py", line 92, in on_adapter_removed
    if path == self.adapter.get_object_path():
AttributeError: 'NoneType' object has no attribute 'get_object_path'
_________
enumerate_connections (/usr/lib/python3/dist-packages/blueman/plugins/applet/ShowConnected.py:53)
Found 0 existing connections 
mike@ITLT-UBUNTU6500:~$ 

* [ *] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

Huckleberry582, 2018-03-21:
By the way, I'm new to this and didn't see the part about posting in patebin or GitHub Gist. Please forgive this noob.

Huckleberry582, 2018-03-22:
I gave a try to downloading the current master but when I tried to use gdebi to install it I was told it was the same as the currently installed package. I tried to reinstall it but was given this output:

dpkg: error: unable to read filedescriptor flags for <package status and progress file descriptor>: Bad file descriptor
I then tried using terminal to install because I read that sometimes gdebi might have issues with a file. I ran dpkg and then apt install with the -f argument per what i read about that error message. I then tested and it didn't correct the issue. Please let me know what else I might try. This is a computer I set up to try out 18.04 on and I am happy to experiment if it will help the community.

Mike

infirit, 2018-03-27:
I installed the latest xubuntu 18.04 beta in a vm and I do not see the high cpu usage. If you open a task/process viewer is it blueman-applet that's at 100% or something else like bluetoothd?

I updated the installation docs for the git snapshots and hopefully this helps you try them out. Make sure you use the pythom 3.6 section.

Huckleberry582, 2018-03-28:
Well, I may be in the wrong place. I assumed that the issue was with Bluetooth just because turning that off seemed to fix the CPU usage. I am attaching a screenshot so you can see what HTOP showed ast the offending process and what the system manager said.
[image]
I'll have to look and see what systemd-udevd does and how it affects bluetooth.

Mike

infirit, 2018-03-29:
udev manages device nodes in /dev when devices are added and removed. It may very well be that it is related to bluetooth just not blueman. You can use ``udevadm monitor`` to see what it is doing when you turn bluetooth on or off.

My suspicion is that udev runs a script or command that behaves badly.

Huckleberry582, 2018-04-04:
I installed UbuntuMate 17.10 as a dual boot to see if it had the same issue and it didn't. But, as soon as I loaded up 18.04, it happened again. Somewhere along the way, however, something got messed up and the thing wouldn't boot either. I ended up just wiping it out and giving mx17.1 a try. It runs beautifully on this old laptop. I think I'll stick with it on this machine for a bit and wait a little while on trying 18.04 again.

Thanks for your help.

closed 2018-04-04 by Huckleberry582