blueman-project/blueman #854: I cannot find any device using blueman

MrBloksa, 2018-03-25:
blueman: 2.0.4
BlueZ: 5.37-0ubuntu5.1
Distribution: Linux Mint 18.3 "Sylvia"
Desktop environment: Cinnamon 3.6

* [ X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

Please I need help with my Bluetooth adaptor, using Bluetooth manager I cannot find any nearby device and if I found one it is impossible to connect, I've tried every option I've found looking in this Troubleshooting page but none of the options have been helpful.

cschramm, 2018-03-26:
> I cannot find any nearby device and if I found one

So do you find one or not? :thinking:

> it is impossible to connect

The device might require pairing first.

You can try to pair the device in `bluetoothctl` and check the result. If it's not helpful, you can try running bluetoothd manually with the `-d` switch to get debugging output in your syslog.

I expect some issue with your hardware (far out of blueman's scope). What chip are you using?

MrBloksa, 2018-03-26:
Hello, thank you very much for the quick reply, I was able to discover new devices and pair them, but afterwards I had connection problems, so I purge the bluez tool and reinstalled it and now I can't even find new devices.

I'm using an HP Stream 11 with intel inside, for more info I ran `sudo lshw` and this is the output: 

``` 
XXXX-hp-stream-XXXXXX-11
    description: Notebook
    product: HP Stream Notebook PC 11 (K2L95UA#ABA)
    vendor: Hewlett-Packard
    version: Type1 - ProductConfigId
    serial: 5CD4517PD4
    width: 64 bits
    capabilities: smbios-2.7 dmi-2.7 vsyscall32
    configuration: boot=normal chassis=notebook family=103C_5335KV G=N L=CON B=HP S=PAV X=Null sku=K2L95UA#ABA uuid=35434434-3531-3750-4434-345031354435
  *-core
       description: Motherboard
       product: 8023
       vendor: Hewlett-Packard
       physical id: 0
       version: 54.11
       serial: PESVH018J7SCUW
       slot: Type2 - Board Chassis Location
     *-firmware
          description: BIOS
          vendor: Insyde
          physical id: 0
          version: F.07
          date: 11/28/2014
          size: 64KiB
          capacity: 4544KiB
          capabilities: pci upgrade shadowing cdboot bootselect edd int9keyboard int14serial int17printer int10video acpi usb zipboot biosbootspecification netboot
     *-cpu
          description: CPU
          product: Intel(R) Celeron(R) CPU  N2840  @ 2.16GHz
          vendor: Intel Corp.
          physical id: 4
          bus info: cpu@0
          version: Intel(R) Celeron(R) CPU  N2840  @ 2.16GHz
          slot: CPU 1
          size: 1306MHz
          capacity: 2582MHz
          width: 64 bits
          clock: 83MHz

```
I also try to discover new devices using `bluetoothctl` but it doesn't work and I don't have the knowledge to run bluetoothd manually with the `-d` switch.

How can I find out if it's a software or a hardware issue?

closed 2018-06-16 by cschramm