blueman-project/blueman #858: Failed to apply network settings Exception: dnsmasq failed to start. Check the system log for errors

rijotech, 2018-04-10:
I'm in Ubuntu 14.04 i developed a signage system on a low end computer i cannot upgrade distro and notice that when I launch blueman-manageris trying to start a new instance of dnsmasq, but dnsmasq  is already running in the background, I did updated the policy file as instructed on a previous issue post but still the same issue. 
Any ideas on how to resolve this issue?
Another thing if i kill the dnsmasq before blueman-manager runs it works fine but then my other services such as google remote stop from working.

rijotech, 2018-04-10:
Never mind i just removed dnsmasq from /etc/init.d and setup blueman-manager to auto start on boot. It works just fine now.

closed 2018-04-10 by rijotech

infirit, 2018-04-10:
>Never mind

Sorry no :smile:. AFAIK dnsmasq will bind to all interfaces by default. You have to force the one from /etc/init.d to bind to a specific interface and you should be good. See [1] what options we pass to it.

[1] https://github.com/blueman-project/blueman/blob/2-0-stable/blueman/main/NetConf.py#L53