blueman-project/blueman #859: [Patch] blueman-applet auto-power-on should respect /etc/bluetooth/main.conf

GM-Script-Writer-62850, 2018-04-14:
blueman: 2.0.4-1ubuntu3
BlueZ: 5.46-0ubuntu3
Distribution: Xubuntu 17.10
Desktop environment: Xfce 4.12

The auto-power-on options should respect the AutoEnable or InitiallyPowered option in the main bluetooth config file to prevent confusion when trying to enable/disable bluetooth at boot

[blueman.patch.zip](https://github.com/blueman-project/blueman/files/1910336/blueman.patch.zip)
This patch expands on commit 8049a9edeb10a9d2fe6751af2a86a6c2b8c8cee7

* Also fixed minor performance issue, no need to set self.adapter_state before if we are about to set it again

infirit, 2018-04-14:
I don't see any confusion. The point of autopower option is to help users deal with the default in bluez5, not powering the adapter. If one does not want blueman to do this then change the setting on the powermanager plugin. If one changes the default in main.conf to power the adapter autopower does not need to do anything.

GM-Script-Writer-62850, 2018-04-14:
I am the person that did not know there were settings till i went into the source code
I had manually set it to false in the main.config file and could not figure out why it was turning on
i had even made a script just to turn it back off one the applet turns it on
said script no longer works due to a [permissions issue](https://askubuntu.com/questions/924344/using-bluetooth-command-without-root-access-in-17-04)
since i decided nobody was going to fix the 'bug' i decided to do it myself
I figured the issue was for some stubborn piece of hardware that did not care if it was not enabling and required extra effort
at first i tried changing the default value of the setting and it did not do what i wanted
i figured the code was checking if the property existed instead of the property value
after going through the xfce4 settings then the gconf settings i went to the code again and saw it was dconf and installed the dconf editor to change the setting, i only just found out last night that there was a settings editor in the applet
me a few days ago: https://ubuntuforums.org/showthread.php?t=2389057

It may be better to use the option to be auto, force on, or force off.
The distributions maintainers can turn bluetooth on by default
i still think it should respect the system configs for bluetooth by default
i would say just drop the settings all together and let the system handle it, but on a multi user system the non admin users should be able to change the setting for there account

this is something that has been bugging my since i upgraded from xubunut 16.04 (was never happy with it after my upgrade from 14.04; needed a clean install *blames systemd*)

cschramm, 2018-04-15:
> what was the 1st ubuntu release to receive commit 8049a9e?

15.10.

GM-Script-Writer-62850, 2018-04-15:
Then I have been working around this feature i did not realize was a feature since 16.04 (2 years)
Can I ask why there is a force option?

Assuming auto-power-on exist only to override the system default, i think it should offer force off, force on, and system default; default should be the system default as overriding a system wide configuration should be opt in

infirit, 2018-04-15:
>Then I have been working around this feature i did not realize was a feature since 16.04 (2 years)

Maybe in the future you ask for help here or on a forum and not forcefully workaround a problem like this.

>Can I ask why there is a force option?

The reason I gave already. The default between bluez 4 and 5 changed to not powering the adapter. This made many users sad so @cschramm added the feature and made them happy.

>i think it should offer force off, force on, and system default

Force on is what it does already and system default is the same as disabling the option in the plugin settings. That leaves force off and I do not see how having it is going to be useful for anyone.

99.9% of the user will want blueman to power the adapter for them. If the .1% want to change it they can in the plugin settings.

GM-Script-Writer-62850, 2018-04-15:
About the force option I meant the `force=True` part
[blueman-applet.patch.zip](https://github.com/blueman-project/blueman/files/1912324/blueman-applet.patch.zip)
Note the dprint message makes it obvious there is a setting
```
auto = self.get_option("auto-power-on")
dprint("D-Bus Option:",
    "\n\t",
    self.__gsettings__['schema'],
    ":",
    "auto-power-on",
    "=",
    auto
)
```
If I had seen something in the console output like that, I likely would have never had a problem

Patch Details (based on original code):

1. Add option to force bluetooth off
2. Add debug message showing option type, location, and value
3. Change default to system default

Maybe somebody will use point one
I see no reason point two should not be added
I see point point three is rejected, but there anyway

closed 2019-04-27 by stale[bot]

lejondev, 2019-09-16:
So how to disable it? I can not find the plugin options at all :(

cschramm, 2019-09-16:
@lejondev: Right click the tray icon, click "Plugins", select "PowerManager" and click the "Settings" button.

lejondev, 2019-09-16:
hmm. That did not work.
There is only a "Configuration" button for me, disabling the the option "Auto power-on _(Automatically power on adapters)_", does not do aything for me it seems (the option stays off after restart). Disabling the whole "PowerManager" plugin (although there are depency issues with KillSwitch) does not help either (this one reenables on startup). 

Also it is weird/unituitive to find the setting here.

[image]

lejondev, 2019-09-16:
Alright. It stays off now.
What I needed to do was [this](https://askubuntu.com/a/155886) as well.