blueman-project/blueman #861: Keyboard shortcut to close window

ranjan-purbey, 2018-04-17:
blueman: 2.0.4
BlueZ: 5.46
Distribution: Ubuntu 17.10
Desktop environment: Mate
Neither Ctrl+W nor Ctrl+Q closes the blueman-manager window.
* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

cschramm, 2018-04-17:
Alt+F4 is a common convention to close a window and works fine for me.

infirit, 2018-04-21:
I am curious where you got those shortcuts from.

closed 2019-04-27 by stale[bot]