blueman-project/blueman #862: Blueman crashed after upgrading to kernel 4.15.14

NeyzanRumi, 2018-04-21:
**This issue persists even after upgrading to Linux Kernel 4.15.17.**
_**Original reports at [LXQt's GitHub page](https://github.com/lxqt/lxqt/issues/1472), [RedHat's Bugzilla](https://bugzilla.redhat.com/show_bug.cgi?id=1564888) and [LinuxQuestions](https://www.linuxquestions.org/questions/fedora-35/blueman-and-added-widgets-in-lxqt-panel-crashed-after-upgrading-to-kernel-4-15-14-a-4175627252/).**_

-------------

blueman: 2.1
BlueZ: 5.49.1
Distribution: Fedora 27
Kernel: 4.15.17-300.fc27.x86_64 SMP
Desktop environment: LXQt 0.11.1
Qt: 5.9.4
Adapter: Qualcomm Atheros Communications AR3012 Bluetooth 4.0
Network Controller: Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)

-------------

* [√ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

-------------

   **_Original post of two weeks ago:_**
After upgrading the kernel to 4.15.14 and restarting the system I found all widgets I added in the panel removed (but not colour and style personalisation) and that Blueman stopped functioning. In blueman's screen no devices were showing and the "Adaptor" and "Device" options in the toolbar were greyed out.
Other personal settings, such as recent documents in LibreOffice or browser history, were not affected.
Also, after re-adding the lost widgets to the panel in 4.15.13 they remained even after rebooting into 4.15.14, but Blueman again stopped functioning.

-------------

   **_Update after upgrading to Linux Kernel 4.15.17:_**
**Issue persists**

-------------

Debug output:

```
[root@192 ~]# blueman-applet --loglevel debug
blueman-applet version 2.1.alpha1 starting
blueman-applet 12.58.03 INFO     PluginManager:85 Load      : ['KillSwitch', 'NetUsage', 'ExitItem', 'PPPSupport', 'DhcpClient', 'StatusIcon', 'ShowConnected', 'AuthAgent', 'SerialManager', 'NMDUNSupport', 'DiscvManager', 'Networking', 'GameControllerWakelock', 'PowerManager', 'NMPANSupport', 'StandardItems', 'RecentConns', 'Menu', 'TransferService', 'DBusService']
blueman-applet 12.58.03 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 12.58.04 WARNING  PluginManager:146 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 12.58.04 WARNING  PluginManager:146 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 12.58.04 INFO     Networking:31 load_nap_settings: Loading NAP settings
blueman-tray version 2.1.alpha1 starting
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 12.58.04 INFO     PluginManager:155 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 12.58.04 INFO     KillSwitch:104 io_event  : killswitch registered 1
blueman-applet 12.58.04 INFO     KillSwitch:120 io_event  : State: True
blueman-applet 12.58.04 INFO     PowerManager:169 UpdatePowerState: off False | foff False | on True | current state True | new state True
blueman-applet 12.58.04 INFO     KillSwitch:104 io_event  : killswitch registered 2
blueman-applet 12.58.04 INFO     KillSwitch:120 io_event  : State: True
blueman-applet 12.58.04 INFO     PowerManager:169 UpdatePowerState: off False | foff False | on True | current state True | new state True
blueman-applet 12.58.04 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 12.58.04 INFO     Applet:47 _on_dbus_name_appeared: org.bluez :1.0
blueman-applet 12.58.04 INFO     BluezAgent:78 register_agent: Register Agent
blueman-applet 12.58.04 ERROR    PluginManager:209 Run       : Function on_manager_state_changed on DiscvManager failed
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/blueman/main/PluginManager.py", line 206, in Run
    ret = getattr(inst, func)(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py", line 47, in on_manager_state_changed
    self.init_adapter()
  File "/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py", line 67, in init_adapter
    self.adapter = self.parent.Manager.get_adapter()
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Manager.py", line 85, in get_adapter
    raise DBusNoSuchAdapterError("No adapter(s) found")
blueman.bluez.errors.DBusNoSuchAdapterError: No adapter(s) found
blueman-applet 12.58.04 INFO     Networking:68 set_nap   : set nap False
blueman-applet 12.58.04 INFO     RecentConns:97 initialize: rebuilding menu
blueman-applet 12.58.04 INFO     ShowConnected:38 enumerate_connections: Found 0 existing connections
blueman-applet 12.58.04 INFO     TransferService:193 _on_dbus_name_vanished: org.bluez.obex not running or was stopped
blueman-applet 12.58.05 INFO     PowerManager:120 RequestPowerState: Requesting True
blueman-applet 12.58.05 INFO     KillSwitch:136 on_power_state_change_requested: True
blueman-applet 12.58.05 DEBUG    KillSwitch:149 on_power_state_change_requested: Using mechanism to set state: True
blueman-applet 12.58.05 INFO     PowerManager:105 check     : callbacks done
blueman-applet 12.58.05 INFO     PowerManager:76 set_adapter_state: True
blueman-applet 12.58.05 INFO     PowerManager:169 UpdatePowerState: off False | foff False | on True | current state True | new state True
```

------------

Furthermore, "bluetoothctl list" gave no output, although Airplane mode is off (other networking methods function normally).

------------

Running "bluetoothctl":

```
[bluetooth]# pair
Missing device address argument
[bluetooth]# head -1 $(which blueman-applet)
[bluetooth]#
```

infirit, 2018-04-21:
The TraceBack in the blueman log is related but not the cause of the problem. The problem is likely with the kernel module responsible for the blutooth device. Here are a couple of things you can check and try.

* Search the boot log for ``Bluetooth`` and ``hci``. If the module is not loaded or not loaded properly it should show there. I think on systemd you run as root ``journalctl bluetooth.service``.
* Make sure you have the firmware installed for the device. The log will tell you if it is missing and/or failed to load.
* Run ``hciconfig`` and see if the device show up.

NeyzanRumi, 2018-04-22:
"journalctl" gave this related output:
`Apr 22 11:36:15 localhost.localdomain kernel: Bluetooth: hci0: don't support firmware rome 0x31010000`

"hciconfig" gives me:
```
hci0:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN 
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:0 acl:0 sco:0 commands:0 errors:0
```

"hciconfig hci0 reset":
`Can't init device hci0: Input/output error (5)`

------------------

I found a .deb file of the latest firmware for Atheros and converted it into .rpm, but I couldn't manage to install it - it conflicts with components of the linux-firmware package. It seems that the problem occurs somewhere there, but I haven't managed yet to find out how exactly.

infirit, 2018-04-22:
Ok, it appears to be a kernel bug, see [1]. Unless you are willing (and able) to patch your kernel I suggest to go back to the previous kernel. The previous kernel should still show up in the boot menu AFAIK.

[1] https://bugzilla.kernel.org/show_bug.cgi?id=199271

NeyzanRumi, 2018-04-22:
I also came to the conclusion that the issue originates from the kernel. I commented and linked to this bug report in the kernel bug report you attached.

BT functions normally only up until 4.15.13, so I reverted back to it and will wait a while to see if this will be solved in later kernel updates.

NeyzanRumi, 2018-04-22:
Just saw the edit. The linux-firmware package does have BT, which seems to function normally with a 4.15.13 kernel but not with later ones. I'll fiddle around a bit and see what comes up...

NeyzanRumi, 2018-04-26:
Issue also reported [here](https://bugzilla.redhat.com/show_bug.cgi?id=1568911).

infirit, 2018-04-27:
OK, closing now. Feel free to add to the issue if you think it may be relevant to others.

closed 2018-04-27 by infirit