blueman-project/blueman #864: duplicate icon

vzaliva, 2018-04-28:
blueman-applet: 2.0.5
Distribution: Ubuntu Bionic (18.04 LTS)
Desktop environment: i3

After upgrading to Ubuntu 18.04 my blueman icon shows twice in the status bar.
I do not see any duplicate processes:

```
root@nemo:~# ps ax| grep -i blue | grep -v grep
  914 ?        Ss     0:00 /usr/lib/bluetooth/bluetoothd
 3044 ?        Ss     0:00 /usr/lib/bluetooth/obexd
30250 pts/1    Sl     0:02 /usr/bin/python3 /usr/bin/blueman-applet
root@nemo:~# 

```

Screenshot:

[image]

cschramm, 2018-04-28:
Looks like two different icons to me. What comes up if you (right- | left-) click on them?

vzaliva, 2018-04-28:
They both bring blueman-applet menu. Selecting 'Help' in it shows the same version (2.0.5). If I kill 'blueman-applet' process both disappear. If I restart it, both re-appear.

vzaliva, 2018-05-04:
When I toggle checkbox for this plugin both icons blink. Regardless of the stat of checkbox both are shown.

cschramm, 2018-05-04:
:thinking: Are you using numix-icon-theme? You might be experiencing issues related to https://github.com/numixproject/numix-icon-theme/issues/1235 then. Please try a different icon theme.

vzaliva, 2018-05-04:
I am not sure what icon theme I am using. I never customized it and I guess it is some default one. How can I check which one it is and how to change it?

cschramm, 2018-05-04:
https://unix.stackexchange.com/questions/150853/how-to-set-gtk-icon-theme-in-i3 should help with that

vzaliva, 2018-05-04:
Using `lxappearance` I've switched to *GNOME* icon theme and it helped. I now have a single blue icon instead of 2 black and white icons. *HighContrast* theme gives you a single black and white icon. *Ubuntu-Mono-Dark* shows double icons as originally reported.

closed 2018-05-04 by vzaliva

cschramm, 2018-05-05:
Yup, looks like Ubuntu-Mono-Dark has the same issue numix had (i.e. non-rectangular (22 px height, but smaller width) Bluetooth icons. I guess i3 just repeats the icon in that case to fill the rectangular box (probably other DEs don't, so that issue does not show often).

Upstream is at https://launchpad.net/ubuntu-themes.