blueman-project/blueman #869: Blueman 2.1alpha2 gentoo issue

gfernandop, 2018-04-28:
blueman: 2.1alpha2
BlueZ: 5.49
Distribution: Gentoo Linux
Desktop environment: XFCE

I had two adapters, since I removed one of them (hci1) by removing the kernel module I get the message pasted below:

Connection Failed: 'NoneType' object is not iterable

It seems blueman keeps hci1 as an adapter and when I try to use my main device (hci0) it crashes.

I have gathered the traces by using --loglevel debug:

> --- Logging error ---
Traceback (most recent call last):
  File "/usr/lib64/python3.5/site-packages/gi/overrides/Gio.py", line 134, in __async_result_handler
    ret = obj.call_finish(result)
GLib.GError: g-io-error-quark: GDBus.Error:org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib64/python3.5/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib64/python3.5/site-packages/blueman/main/DbusService.py", line 33, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib64/python3.5/site-packages/blueman/plugins/applet/DBusService.py", line 61, in connect_service
    self.parent.Plugins.RecentConns.notify(object_path, uuid)
  File "/usr/lib64/python3.5/site-packages/blueman/plugins/applet/RecentConns.py", line 199, in notify
    for i in self.items:
TypeError: 'NoneType' object is not iterable
 (36)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.5/logging/__init__.py", line 986, in emit
    msg = self.format(record)
  File "/usr/lib64/python3.5/logging/__init__.py", line 836, in format
    return fmt.format(record)
  File "/usr/lib64/python3.5/logging/__init__.py", line 573, in format
    record.message = record.getMessage()
  File "/usr/lib64/python3.5/logging/__init__.py", line 336, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/usr/bin/blueman-manager", line 47, in <module>
    Gtk.main()
  File "/usr/lib64/python3.5/site-packages/gi/overrides/Gio.py", line 139, in __async_result_handler
    error_callback(obj, e, real_user_data)
  File "/usr/lib64/python3.5/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 182, in fail
    logging.info("fail", result)
Message: 'fail'
Arguments: (GLib.Error('GDBus.Error:org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib64/python3.5/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib64/python3.5/site-packages/blueman/main/DbusService.py", line 33, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib64/python3.5/site-packages/blueman/plugins/applet/DBusService.py", line 61, in connect_service
    self.parent.Plugins.RecentConns.notify(object_path, uuid)
  File "/usr/lib64/python3.5/site-packages/blueman/plugins/applet/RecentConns.py", line 199, in notify
    for i in self.items:
TypeError: 'NoneType' object is not iterable
', 'g-io-error-quark', 36),)

infirit, 2018-04-28:
Does it happen every time you start or only when you remove the module? Also please post the whole log not just the error.

gfernandop, 2018-04-29:
I have discovered it fails only when running as normal user:
https://paste.pound-python.org/show/oZZ5g6skzmSfnQCTgiwd/

Here you can see t

gfernandop, 2018-04-29:
Executing blueman-applet and then blueman-manager --loglevel debug:
https://paste.pound-python.org/show/TjkXefZvEpsyQPTkSSDD/

gfernandop, 2018-04-29:
Unintentionally closed it, just reopening.

infirit, 2018-04-29:
Yes running as another user will fix this. The problem is that we store the default adapter in GSettings and it will try to use it on next startup. I have changed the code a few time and I messed up (twice now).

Can you try #870 and see if the problem goes away.

gfernandop, 2018-04-30:
Thanks for your support.

Does git have a way to apply the changes and revert if needed?
I work as network analyst having basic git experience.
Should I use a git command to apply the patch or replace each file manually (keeping a backup)?

Best Regards

infirit, 2018-04-30:
@gfernandop if you are building blueman-9999 download https://patch-diff.githubusercontent.com/raw/blueman-project/blueman/pull/870.patch and stick it in ``/etc/portage/patches/net-wireless/blueman/``. Portage will apply the patch now when you emerge blueman.

infirit, 2018-04-30:
Ok so it does not error anymore, right? I'll merge the changes to master.

If you go back to 2.1-alpha2 you can reset the key with ``gsettings reset org.blueman.general last-adapter`` to fix the error.

I did notice the NoneType error as well and I'll look into that.

gfernandop, 2018-04-30:
It is strange because when I try to connect, the first time shows me the error.
If exit by clicking the applet "Exit" button and re-open it the error is gone.
I can confirm if the problem really is solved, do you understand?

infirit, 2018-04-30:
Sorry, I am being confusing :disappointed:. There is a bug that causing the TraceBack [1] and that should be fixed now. I was hoping this would fix the NoneType error when trying to connect. 

If the NoneType error persists after the patch and the TraceBack below [1] is gone then I'll have to look further.

[1] TraceBack
```
blueman-manager 22.25.13 ERROR    DeviceList:246 SetAdapter: No adapters found with pattern: hci1
Traceback (most recent call last):
  File "/usr/lib64/python3.5/site-packages/blueman/gui/DeviceList.py", line 236, in SetAdapter
    self.Adapter = self.manager.get_adapter(adapter)
  File "/usr/lib64/python3.5/site-packages/blueman/bluez/Manager.py", line 91, in get_adapter
    raise DBusNoSuchAdapterError("No adapters found with pattern: %s" % pattern)
blueman.bluez.errors.DBusNoSuchAdapterError: No adapters found with pattern: hci1
```

gfernandop, 2018-04-30:
Me too.
When should I use "gsettings reset org.blueman.general last-adapter"? version 9999+patch or 2.1alpha2?

Now my system has version 9999 + patch.
I rebooted the system and opening the application I got the same error.

gfernandop, 2018-04-30:
What does mean "key"? Clean it using gsettings?
Could I remove my old hci1 device and keeps only hci0?
In POV of blueman because my system does not list hci1 anymore.

gfernandop, 2018-04-30:
If needed I can use each version and gather the logs using or not the patch and executing the gsettings command.
Just enumerate the steps and I can follow them strictly aiming to help you.

infirit, 2018-04-30:
Just tested the changes and it fixes all the problems. I merged to master so just remove the patch from /etc/portage/patches and emerge blueman-9999 again. You should see no problems anymore.

gfernandop, 2018-04-30:
Did you try to disconnect and re-connect?
The same error message appears...

gfernandop, 2018-04-30:
Seems to be fixed.
Thanks for you support :+1: