blueman-project/blueman #873: ModuleNotFoundError: No module named 'blueman'

krichter722, 2018-05-03:
blueman: 2.1.alpha2-80-gc67ae920
BlueZ: bluez
Distribution: Ubuntu 18.04
Desktop environment: MATE

* [ x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

After a source installation with `./autogen.sh && ./configure && make && sudo make install` invoking `blueman-manager` fails due to

``` 
Traceback (most recent call last):
  File "/usr/local/bin/blueman-manager", line 18, in <module>
    from blueman.main.Manager import Blueman
ModuleNotFoundError: No module named 'blueman'
```

The system package `blueman` has been removed. Other `ImportError`s can be found at https://travis-ci.org/krichter722/blueman/jobs/374330197. This might indicate that more test coverage on the CI service is needed.

cschramm, 2018-05-03:
Debian's (and thus probably Ubuntu's) system python version does not include any `site-packages` path in `sys.path` by default, so the modules does not get picked up from the path you're installing it into (probably `/usr/local/lib/python3.6/site-packages`). You can either add the path when running (`PYTHONPATH=/usr/local/lib/python3.6/site-packages`) or install as a package letting debhelper change things from `site-packages` to `dist-packages`.

I'm not aware of straightforward way to let `make install` put the module into `/usr/lib/python3/dist-packages` (for obvious reasons).

cschramm, 2018-05-03:
By the way: `./autogen.sh && ./configure` does not make much sense as `autogen.sh` by default invokes `configure`.

cschramm, 2018-05-03:
There's another way: If `~/.local/lib/python3.6/site-packages` does exist, it gets added to the path, so configuring the respective prefix (`~/.local`) to install it there should make things work out of the box.

infirit, 2018-05-03:
Even if you solve instalation there is nothing to be gained by running blueman-manager under travis. Travis does not have a bluetooth device and also it would block forever as we can not interact with it. If you are interested time is best spend getting test added to blueman, see #43

krichter722, 2018-05-03:
> Even if you solve instalation there is nothing to be gained by running blueman-manager under travis. 

This is just a demonstration of import errors. I didn't want to suggest to run `blueman-manager` inside Travis and hope I didn't.

cschramm, 2018-05-03:
I can confirm that `--prefix="$HOME/.local" --datadir="/usr/share"` results in an installation that works fine with the system's python on Debian.

closed 2018-05-03 by cschramm

philippludwig, 2018-08-27:
Yeah this should go definitely into the readme that you need to pass ``--prefix=/usr`` to get a working installation.