blueman-project/blueman #875: start of blueman-applet fails due to "GLib.Error: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.152 was not provided by any .service files (2)"

krichter722, 2018-05-04:
blueman: 2.1.alpha2-80-gc67ae920
BlueZ: 5.48-0ubuntu3
Distribution: Ubuntu 18.04
Desktop environment: MATE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

After installing `blueman` with `./configure --prefix="$HOME/.local" --datadir="/usr/share" && make && sudo make install`, invoking `~/.local/bin/blueman-applet --loglevel debug` fails due to

```
blueman-applet version 2.1.alpha2 starting
Stale PID, overwriting
blueman-applet 07.55.42 INFO     PluginManager:85 load_plugin: ['AppIndicator', 'AuthAgent', 'DBusService', 'DhcpClient', 'DiscvManager', 'ExitItem', 'GameControllerWakelock', 'KillSwitch', 'Menu', 'NMDUNSupport', 'NMPANSupport', 'NetUsage', 'Networking', 'PPPSupport', 'PowerManager', 'RecentConns', 'SerialManager', 'ShowConnected', 'StandardItems', 'StatusIcon', 'TransferService']
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 07.55.42 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Address': '68:17:29:77:05:46', 'AddressType': 'public', 'Name': 'richter-Lenovo-IdeaPad-Z500', 'Alias': 'richter-Lenovo-IdeaPad-Z500', 'Class': 1835276, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001112-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001133-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0530'}
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 07.55.42 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 07.55.42 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 07.55.42 INFO     Networking:31 load_nap_settings: Loading NAP settings
blueman-tray version 2.1.alpha2 starting
Stale PID, overwriting
blueman-tray version 2.1.alpha2 starting
There is an instance already running
blueman-applet 07.56.07 ERROR    BasePlugin:76 _load     : Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms)
Traceback (most recent call last):
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
GLib.GError: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms) (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/plugins/BasePlugin.py", line 70, in _load
    self.on_load()
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 24, in on_load
    self.load_nap_settings()
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/plugins/applet/Networking.py", line 45, in load_nap_settings
    m = Mechanism()
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 15, in __call__
    cls._instance = super().__call__(*args, **kwargs)
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 44, in __init__
    systembus=True, *args, **kwargs)
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 38, in __init__
    raise DBusProxyFailed(e.message)
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms)
blueman-applet 07.56.07 ERROR    PluginManager:161 __load_plugin: Failed to load Networking
Traceback (most recent call last):
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/main/DBusProxies.py", line 36, in __init__
    self.init()
GLib.GError: g-dbus-error-quark: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms) (20)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/main/PluginManager.py", line 159, in __load_plugin
    inst._load()
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/plugins/AppletPlugin.py", line 47, in _load
    super(AppletPlugin, self)._load()
[... 43 lines cut ...]
blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.blueman.Mechanism': timed out (service_start_timeout=25000ms)
blueman-applet 07.56.07 DEBUG    BasePlugin:43 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Networking.Networking object at 0x7f0a7c1b9e10>
blueman-applet 07.56.07 DEBUG    BasePlugin:43 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.NMPANSupport.NMPANSupport object at 0x7f0a7c1b9da0>
blueman-applet 07.56.07 DEBUG    BasePlugin:43 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.NMDUNSupport.NMDUNSupport object at 0x7f0a7c1b9ef0>
blueman-applet 07.56.07 DEBUG    BasePlugin:43 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.KillSwitch.KillSwitch object at 0x7f0a7c1b9ac8>
blueman-applet 07.56.07 DEBUG    BasePlugin:43 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.PowerManager.PowerManager object at 0x7f0a7c1b99b0>
blueman-applet 07.56.07 DEBUG    BasePlugin:43 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock object at 0x7f0a7c1b9668>
blueman-applet 07.56.07 DEBUG    BasePlugin:43 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.ExitItem.ExitItem object at 0x7f0a7c1b9518>
[... cut ...]
  File "/home/richter/.local/bin/blueman-tray", line 40, in <module>
    BluemanTray()
  File "/home/richter/.local/lib/python3.6/site-packages/blueman/main/Tray.py", line 23, in __init__
    indicator_name = applet.GetStatusIconImplementation()
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 191, in __call__
    None)
GLib.Error: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.152 was not provided by any .service files (2)
```

Previous instances of `blueman-applet` have been killed with `killall`.

cschramm, 2018-05-04:
Looks like I didn't remove the configuration files when I tested for #873, sorry.

As your prefix is `$HOME/.local`, `org.blueman.Mechanism.conf` is not where your system looks for it. Add `--sysconfdir=/etc`.

Note that I do provide a [deb package of the git master](https://github.com/blueman-project/blueman/wiki/Packaged-versions#debian-gnulinux) to save you from having to know your system's and automake's details.

cschramm, 2018-05-04:
Gnar, the dbus daemon cannot start blueman-mechanism as it does not include the `site-packages` path of course. You can start it manually with `sudo -E ~/.local/libexec/blueman-mechanism` and start `blueman-applet` while it's running.

You cannot build from source for the system python without the hassle with include paths, sorry.

closed 2019-04-27 by stale[bot]