blueman-project/blueman #877: Promote auto-power setting

cschramm, 2018-05-15:
The PowerManager plugin has a setting to automatically enable bluetooth as that is useful to some users. On the other hand every once in a while users are irritated by that and do not find the setting to disable it.

One idea to improve that: Ask the user about his preference on first start (maybe providing the way to get to the setting to change it later, but he will forget anyway :wink:).

Another idea: Move all settings to a single dialog. It's much easier to find one's way there instead of the PowerManager plugin settings. Regarding plugins I'm only counting four settings. Might make sense to join them with local services as well, though.

sixtyfive, 2018-05-15:
My expectation, upon right-clicking the applet, was to find a menu item "Settings". Instead, there were "Devices", "Adaptors", "Local Services" and "Plug-ins", non of which seemed to bring me further. If you ask about the preferred UI, from a user's perspective, I think it could look somewhat like the Gnome or Budgie settings panels, where on the left you have a pane with different sections and on the right the settings for the selected section. The applet's context menu would only have the "Settings" item in it.

infirit, 2018-05-15:
>Ask the user about his preference on first start

Yes, a first use Gtk.Assistant should work well.

>Move all settings to a single dialog

Creating a ui around all settings is a good idea and having most of them in GSetings/Dconf makes it pretty easy. But I think it would be good to deal with NetConf if we are going to do this.

With the move to GSettings I tried putting all the settings into it but ran into ip and netmask (maybe also the dhcpd handler, don't remember) stored by NetConf. It is run by mechanism as root and its GSettings do not reflect those of the normal user.

cschramm, 2018-05-16:
A downside of such a first-use mechanism is, of course, that it would pop up without context when a user boots a distribution that ships blueman (most notably Xubuntu, Lubuntu, and Ubuntu MATE) or after installing an additional desktop that includes it (e.g. Cinnamon or MATE on Debian / Ubuntu). I wouldn't expect and want that, both from a user and a maintainer perspective.

I thus guess just making it easier to find the setting but keeping a default (I'm open to discuss if "no auto-power" is a better choice) is the best way to go.

infirit, 2018-05-16:
>A downside of such a first-use mechanism is, of course, that it would pop up without context when a user boots a distribution that ships blueman

True, if we have the applet handle this. However I think it would make more sense to have blueman-manager do this when it is first run. But then we potentially are left with the problem this discussion started with, auto powering the adapter so no auto-power would be needed then.

cschramm, 2018-05-16:
That could be a way...

We could start with no explicit auto-power setting and *not* auto-power in blueman-applet as long as it's not set.
When `blueman-manager` starts and there is no setting, we can ask the user for his preference. At that point we both know that he actually wants to use bluetooth / blueman and that he's currently in that context.

PR #1021 (2019-05-05): Ask user for auto-power-on setting
Closes #877
commit 19c59502 (2019-05-05): Ask user for auto-power-on setting

Closes #877
closed 2019-05-05 by infirit