blueman-project/blueman #878: blueman manager crashed constantly

bilal-hanif43, 2018-05-27:
blueman: version (2.0.5-1+b1).
BlueZ: bluetoothd -v 5.49
Distribution: Linux kali 4.15.0-kali3-amd64 #1 SMP Debian 4.15.17-1kali1 (2018-04-25) x86_64 GNU/Linux
Desktop environment: GNOME Shell 3.28.0

I am using bluetooth-headphones but it crashed when I start blueman-manager
[image]

on PC starup, I am able to connect my headphone for 10-15 minutes only after that it constantly crashed every time

bilal-hanif43, 2018-05-27:
`Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['TransferService', 'ExitItem', 'AuthAgent', 'PPPSupport', 'NMDUNSupport', 'AppIndicator', 'ShowConnected', 'StandardItems', 'Menu', 'DBusService', 'NetUsage', 'DiscvManager', 'PowerManager', 'Networking', 'RecentConns', 'KillSwitch', 'StatusIcon', 'GameControllerWakelock', 'NMPANSupport', 'SerialManager', 'Headset', 'DhcpClient'] 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/obex/Base.py:20)
Detected BlueZ integrated obexd 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
Not loading PPPSupport because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib/python3/dist-packages/blueman/plugins/applet/Networking.py:36)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
[... 40 lines cut ...]
_________
enumerate_connections (/usr/lib/python3/dist-packages/blueman/plugins/applet/ShowConnected.py:53)
Found 0 existing connections 
_________
on_registered (/usr/lib/python3/dist-packages/blueman/bluez/obex/AgentManager.py:18)
/org/blueman/obex_agent 
_________
RequestPowerState (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:131)
Requesting True 
_________
on_power_state_change_requested (/usr/lib/python3/dist-packages/blueman/plugins/applet/KillSwitch.py:118)
True 
_________
UpdatePowerState (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:159)
off False 
foff False 
on True 
current state True 
new state True 
_________
check (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:119)
callbacks done 
_________
set_adapter_state (/usr/lib/python3/dist-packages/blueman/plugins/applet/PowerManager.py:90)
True`

cschramm, 2018-06-01:
> I am using bluetooth-headphones but it crashed when I start blueman-manager

What's `bluetooth-headphones`?

Your screenshot rather suggests that `blueman-manager` freezes. It's still there, right?

`blueman-applet` keeps responding?

Maybe the output from `blueman-manager` gives us a clue what happens. Just start it from the terminal. It might also be useful to kill it once it freezes and get a traceback.

bilal-hanif43, 2018-06-02:
I have beats solo2 wireless headphones. Tried to start `blueman-manager` in terminal and get these logs

_________
on_property_changed (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:155)
adapter propery changed Discovering 0 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
STN-019 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
STN-019 
_________
set_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:85)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f922c0ae1b0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x202eb60)> 
_________
set_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:85)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f922c0b9e58 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x202ee40)> 
_________
on_device_property_changed (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:163)
list: device_prop_ch Connected 1 /org/bluez/hci0/dev_30_21_15_85_AE_88 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Connected 1 
_________
monitor_power_levels (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:185)
starting monitor 
_________
level_setup_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:325)
animating up 
_________
success (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:113)
success  
_________
unset_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:99)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f922c0ae1b0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x202eb60)> 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
STN-019 
_________
unset_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:99)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f922c0b9e58 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x202ee40)> 
_________
__event_callback (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:641)
9 1 
_________
on_pa_event (/usr/lib/python3/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:43)
9 1 
card
add
_________
__event_callback (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:641)
4 26 
_________
on_pa_event (/usr/lib/python3/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:43)
4 26 
_________
__event_callback (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:641)
25 1 
_________
on_pa_event (/usr/lib/python3/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:43)
25 1 
card
change
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
STN-019 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
STN-019 
/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerProgressbar.py:165: Warning: Source ID 657 was not found when attempting to remove it
  GObject.source_remove(self.gsource)
^[	_________
on_device_property_changed (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:163)
list: device_prop_ch ServicesResolved 1 /org/bluez/hci0/dev_30_21_15_85_AE_88 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event ServicesResolved 1 
_________
__event_callback (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:641)
25 1 
_________
on_pa_event (/usr/lib/python3/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:43)
25 1 
card
change
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
STN-019 
_________
__event_callback (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:641)
25 1 
_________
on_pa_event (/usr/lib/python3/dist-packages/blueman/plugins/manager/PulseAudioProfile.py:43)
25 1 
card
change
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:173)
STN-019

cschramm, 2018-06-02:
Looks unsuspicious so far. Then it freezes? No further output when you terminate it (CTRL + C)?

closed 2018-06-16 by cschramm