blueman-project/blueman #880: Bluetooth device detects and disappear quickly

arjunbkool, 2018-05-31:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

arjunbkool, 2018-05-31:
https://forums.linuxmint.com/viewtopic.php?f=90&t=270252&sid=e52f3ede55cd67060fa937e0e06682c7
The issue is described in the linux mint forum. I have found very little help about this issue so far.

cschramm, 2018-06-01:
I'd try re-pairing the device.

arjunbkool, 2018-06-03:
I would like to. But to repair it, the device should be recognised first. and after that, while connecting is when i press the pair button located on my speaker. but unfortunately the device simply blinks out in the detected devices list.

cschramm, 2018-06-04:
Hrm, right. If it vanishes from the list, it's probably not paired (at least not from your PC's point of view; maybe the device itself thinks different).

You can try `remove B8:D5:0B:62:C8:27` anyway. Maybe it helps. Probably not.

The dual boot comment reminds me of those creepy Broadcom chips. The developers of your controller's driver are probably the goto point for further troubleshooting. Or maybe you're lucky and there's an alternative driver that just works for you?

BlueZ' syslog messages might also have some information about what's going on (you can try stopping the bluetooth service and run `bluetoothd` with the `-d` switch manually to get more).

Closing as this is not related to blueman and I don't think we really can help you with your issue. Feel free to continue the thread, though. Maybe we can.

closed 2018-06-04 by cschramm