blueman-project/blueman #881: "Connection Failed: 'NoneType' object is not iterable" when trying to connect to audio device

lukehutch, 2018-06-01:
blueman: blueman-2.1-0.9.alpha2.fc28.x86_64
BlueZ: bluez-5.49-3.fc28.x86_64
Distribution: Fedora 28
Desktop environment: XFCE

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

I have been unable to add Bluetooth audio adapters on Fedora Linux. Here is the blueman-applet debug log on startup:

https://pastebin.com/tDjJCJiE

If I click "Search", right click on a discovered device, then click "Connect", I get the error message "Connection Failed: 'NoneType' object is not iterable". The log is:

https://pastebin.com/cLy1UTns

If instead of clicking "Connect", I click "Setup..." then "Pair", blueman-manager says "Pairing is in progress...", but then it gets stuck. The log is:

https://pastebin.com/856VKxFf

There's no way to close the popup "Pairing" dialog, and control never returns to blueman-manager. If I kill blueman-assistant and blueman-manager, then re-open blueman-manager, then right click on the device and go "Setup..." again, it immediately says "Congratulations, device successfully added". The log says:

https://pastebin.com/RKMDFLDM

However, back in blueman-manager, right-clicking on the device and clicking "Connect" still gives the same NoneType error shown above.

I observe the same problems on two different machines running Fedora (one laptop with built-in Bluetooth, and a desktop with a Bluetooth USB adapter).

Searching for similar "NoneType" errors indicates there have been several errors like this in the past, but not isolated to just one place in the code. I can't figure out how to get a stacktrace from the code that is throwing the "NoneType" error.

infirit, 2018-06-01:
This was fixed with c67ae920c5bfd3932ec6715cc0336139b436f88b. There was a report on this a month ago, see #869.

If you can't apply the patch you can also workaround it by running the below from a terminal.
```
gsettings reset org.blueman.general last-adapter
```

lukehutch, 2018-06-01:
@infirit I wondered why I was seeing an error about an unknown MAC address in thelogs. However, I ran the command you suggested, and it did not fix the problem.

infirit, 2018-06-01:
Yes, there handling of removed adapters got broken by me. There is another place this caused problems. Try

```
gsettings reset org.blueman.plugins.recentconns recent-connections
```

lukehutch, 2018-06-01:
@infirit I just tried both commands in succession, then tried connecting with an already-paired device. Still the same error message.

I suggest putting more debug info into the popup message when an exception is thrown, e.g. source code file and line number, if that is possible...

Other than that though, how do I get a stacktrace for this exception?

infirit, 2018-06-01:
That should have worked around it. To get detailed info run blueman-manager from a terminal,
```
blueman-manager --loglevel debug
```

lukehutch, 2018-06-01:
Here are the logs: https://pastebin.com/Sd7xbvwv

infirit, 2018-06-09:
Ok then there is no way to workaround it. Apply https://github.com/blueman-project/blueman/commit/c67ae920c5bfd3932ec6715cc0336139b436f88b and all will be working as it should.

lukehutch, 2018-06-09:
I applied the patch, and it did fix the bug, thank you. I now have a range of other issues trying to connect audio devices, but I'll file a separate bug. Thanks.

closed 2018-06-09 by lukehutch