blueman-project/blueman #882: Can't connect audio devices

lukehutch, 2018-06-09:
blueman: blueman-2.1-0.9.alpha2.fc28.x86_64 (patched to fix NoneType errors)
BlueZ: bluez-5.50-1.fc28.x86_64
Distribution: Fedora 28
Desktop environment: XFCE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I tried connecting three separate audio speakers / headsets, and all of them failed. I tried a combination of Pair, Connect and Trust, in different orders. The blueman-manager debug logs are attached.

[blueman-manager.log](https://github.com/blueman-project/blueman/files/2087307/blueman-manager.log)

cschramm, 2018-06-09:
Please provide the blueman-applet log and describe what your trying. My guess: You're trying to pair an Anker SoundCore 2 that's in paring mode (i.e. the blue indicator flashes) and get "Authentication Failed". Does BlueZ give you anything in your syslog?

lukehutch, 2018-06-10:
Here's what I did -- refer to the newer blueman-manager.log below (I followed basically the exact same actions, but maybe in a slightly different order) and other logfiles.

# As root:

```
tail -F /var/log/messages
service bluetooth restart
```

As non-root:

```
blueman-applet --loglevel debug &> /tmp/blueman-applet.log &
blueman-manager --loglevel debug &> /tmp/blueman-manager.log &
```

* In blueman-manager, right-clicked on the three existing devices, and selected "Remove"
* Switched on the following three Bluetooth devices, and put them in pairing mode:
  * Tronsmart Element T6
  * SoundCore 2
  * WH-1000XM2
* Clicked "Search" -- the three devices above appear in the list
* Right-clicked on Tronsmart Element T6 in blueman-manager, and clicked "Pair" -- got a desktop popup saying "Pairing failed for Tronsmart Element T6", no reason is given. Tried again, same effect, but after the second time, if I right click on the device again, the top menu option says "Disconnect", so presumably it's connected. The speaker is still blinking blue though, showing it has not yet paired.
* Right-clicked Tronsmart Element T6 and clicked Disconnect, then Connect. Progress bar at bottom right of blueman-manager indicates something is happening, but it sits there for a long time, with no effect. The two other devices are still in pairing mode, but they disappeared from the list of devices in blueman-manager, due to some sort of timeout, presumably. After a couple of minutes, the progress bar freezes, and the Tronsmart device still has not connected. Right-clicking it gives one grayed-out menu item, "Connecting...". There appears to be no way of canceling this pairing operation, and it's stuck. (The little red stop sign icon after the progress bar is grayed out. It would be great if this worked to cancel operations...)
* Clicked Search again, the other two devices pop back up.
* Right-clicked on SoundCore 2, clicked Pair rather than Connect. I get "Connection Failed: input/output error".
* Shortly thereafter, "Connection Failed: Resource temporarily unavailable" pops up. This second failure appears to be the Tronsmart pairing finally failing, since right-clicking gives me the usual menu options again. (N.B. it would be really helpful to list the device name in failure messages like this, for cases where you're dealing with multiple devices!)
* The SoundCore speaker made a dinging sound, indicating pairing half-succeeded, but the LED is still blinking blue, indicating pairing was not completed.
* Clicked Search again, since the WH-1000MX2 headset had disappeared from the list again.
* Right-clicked WH-1000MX2, and selected Trust. A yellow star icon appeared immediately on the headset. Clicked Pair -- no response, but now right-clicking, the menu has changed to say "Disconnect" rather than "Connect". The headset is still blinking blue, indicating is is unpaired and unconnected. Selected Disconnect, then Connect. Same process with progress bar hanging for a couple of minutes as seen with the Tronsmart. (Log snapshot taken before waiting for timeout.)

[blueman-applet.log](https://github.com/blueman-project/blueman/files/2087441/blueman-applet.log)
[blueman-manager.log](https://github.com/blueman-project/blueman/files/2087442/blueman-manager.log)
[dmesg.log](https://github.com/blueman-project/blueman/files/2087444/dmesg.log)
[var_log_messages.log](https://github.com/blueman-project/blueman/files/2087445/var_log_messages.log)

cschramm, 2018-06-10:
The messages in your dmesg.log are really interesting. I expect all other issues to just be a consequence of those issues.

Googling for the messages brings up a bunch of cases.

Issues often relate to USB issues on Raspberry Pi. It uses a custom USB driver so those should not apply to you.

In https://bbs.archlinux.org/viewtopic.php?id=198718 jordi reports he can fix it by reloading the btusb module (so it is related to a USB controller as well), however he describes it as being seldom, so this might not apply to you as well.

In https://bugzilla.redhat.com/show_bug.cgi?id=1521080 there's an issue with several Fedora kernel builds but it should not apply to Fedora 28.

https://forum.manjaro.org/t/bluetooth-crashes-since-updated-to-kernel-4-13-2/31809 discusses issues with BT / WiFi coexistence with Intel chips 8260 and 8265 (in Manjaro Linux). jsbach suggest do disable it in iwlwifi.conf. https://bugzilla.kernel.org/show_bug.cgi?id=197591 also describes a bug in the 8265 firmware. Luca Coelho suggests to remove iwlwifi-8265-34.ucode so that an older version will get loaded or update the kernel to support the newer one. The respective fix landed in Linux 4.14.3. https://bugzilla.kernel.org/show_bug.cgi?id=197147 discusses coexistence issues. I did not extract immediate directions for you from it other than update kernel and firmware.

The list goes on (`"link tx timeout" -raspberry` gives me ~150 results in Google).

So the first thing to do: Check your hardware, Linux version, and firmware versions for your hardware (and of course make sure they are up to date; applying available updates will probably fix the issue).

lukehutch, 2018-06-10:
Thanks for the sleuthing. I have the latest kernel and all the latest Fedora 28 packages:

```
$ uname -a
Linux localhost.localdomain 4.16.13-300.fc28.x86_64 #1 SMP Wed May 30 14:31:00 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

`modprobe -r btusb ; modprobe btusb` does not fix the problem.

The Bluetooth dongle identifies as (via lsusb): `0a5c:200a Broadcom Corp. BCM2035 Bluetooth dongle`, so it's not the Intel firmware issue that you found info on.

`udevadm info -a -p /sys/class/bluetooth/hci0` gives:

```
  looking at device '/devices/pci0000:00/0000:00:07.1/0000:1d:00.3/usb3/3-3/3-3.1/3-3.1:1.0/bluetooth/hci0':
    KERNEL=="hci0"
    SUBSYSTEM=="bluetooth"
    DRIVER==""

  looking at parent device '/devices/pci0000:00/0000:00:07.1/0000:1d:00.3/usb3/3-3/3-3.1/3-3.1:1.0':
    KERNELS=="3-3.1:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="btusb"
    ATTRS{authorized}=="1"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bInterfaceClass}=="e0"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceProtocol}=="01"
    ATTRS{bInterfaceSubClass}=="01"
    ATTRS{bNumEndpoints}=="03"
    ATTRS{supports_autosuspend}=="1"

  looking at parent device '/devices/pci0000:00/0000:00:07.1/0000:1d:00.3/usb3/3-3/3-3.1':
    KERNELS=="3-3.1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="e0"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{bDeviceSubClass}=="01"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="100mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 3"
    ATTRS{bcdDevice}=="0001"
    ATTRS{bmAttributes}=="a0"
    ATTRS{busnum}=="3"
    ATTRS{configuration}==""
    ATTRS{devnum}=="3"
    ATTRS{devpath}=="3.1"
    ATTRS{idProduct}=="200a"
    ATTRS{idVendor}=="0a5c"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Broadcom"
    ATTRS{maxchild}=="0"
    ATTRS{product}=="CCBT2035BDGP23-1"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="removable"
    ATTRS{serial}=="000272C54CE9"
    ATTRS{speed}=="12"
    ATTRS{urbnum}=="12125"
    ATTRS{version}==" 1.10"
```

`bluetoothctl list` shows that a MAC address is assigned to the controller:

```
Controller 00:02:72:C5:4C:E9 localhost.localdomain [default]
```

Support for the BCM2035 seems to have been in the kernel (in bluez) since 2006:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=5094a9971637f88dd38383faa7be81dafd997adb

I found an old mention of the fact that BCM203x firmware could not be included in Fedora, and in fact in other distributions there is a bluez-firmware package containing this firmware:

https://bugzilla.redhat.com/show_bug.cgi?id=849339

However, the firmware specifically refers to (in its filename) BCM2033, not BCM2035. And if my current kernel didn't already have the BCM2035 firmware loaded somehow, presumably I wouldn't be able to run a Bluetooth scan, which I can do successfully. I also found a few references around the Web to people successfully using BCM2035 on Linux (no mention of whether they needed special firmware installed) -- along with many random mentions of it not working...

For good measure though, I put the BCM2033 firmware into /usr/lib/firmware/brcm , and then reloated the btusb module. I don't know if the firmware needs a specific name, or if I need to do something to get the system to pick up the new firmware, but it didn't help.

I guess this is probably not a blueman bug though (and maybe I just need to buy a new adapter), so feel free to close this bug, but thanks for taking a look anyway.

infirit, 2018-06-10:
Traveling ATM..

Missing firmware results in strange problems. For example my brcm USB
dongle will work fine for Bluetooth <4 without firmware but refuses to make
my Bluetooth 4 mouse actually work.

Grep dmesg for the chip and it will show whether it could load the firmware
or not. When I'm back home I'll also have a look.

lukehutch, 2018-06-11:
There's no mention at all of the firmware in dmesg or /var/log/messages. Just the following:

```
[255394.682507] usb 3-3.1: New USB device found, idVendor=0a5c, idProduct=200a
[255394.682510] usb 3-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[255394.682512] usb 3-3.1: Product: CCBT2035BDGP23-1
[255394.682514] usb 3-3.1: Manufacturer: Broadcom
[255394.682515] usb 3-3.1: SerialNumber: 000272C54CE9
[255406.278734] Bluetooth: hci0: command 0x0c14 tx timeout
[255435.200410] Bluetooth: hci0: last event is not cmd complete (0x0f)
[255444.677640] Bluetooth: hci0: command 0x0405 tx timeout
[255455.843456] Bluetooth: hci0: last event is not cmd complete (0x0f)
[255475.959535] Bluetooth: hci0: last event is not cmd complete (0x0f)
[255491.963444] Bluetooth: hci0: last event is not cmd complete (0x0f)
[255507.964449] Bluetooth: hci0: last event is not cmd complete (0x0f)
[255523.999476] Bluetooth: hci0: last event is not cmd complete (0x0f)
[255539.962474] Bluetooth: hci0: last event is not cmd complete (0x0f)
[... lots more of these ...]
```

lukehutch, 2018-06-11:
Another possibility: I can just send you this Bluetooth dongle if you want to play with it. I can buy another one, they're not expensive. It's an old chipset, but a common one, so it would be nice to have this working.

Re. the firmware for BCM203x: as I mentioned, the actual filenames in the bluez-firmware archive specifically refer to BCM2033, not to BCM2035 (which is my USB dongle's chipset). But elsewhere in several places it is stated that this is the firmware for BCM203x. Apparently RedHat thought that they couldn't distribute the firmware, but Debian received explicit permission from Broadcom to distribute this firmware, and some people at RedHat thought that that permission should extend to other Linux distributions too.

lukehutch, 2018-06-11:
No worries, thanks for taking a look.

infirit, 2018-06-11:
For reference this is how my dmesg look like with firmware in place and without. The m720 is the BT 4 mouse I mentioned earlier :smile:.

Notice the line with "Direct firmware load for" trying to find the file and not finding it.

Without
```
[ 1522.646633] usb 1-1.5: new full-speed USB device number 6 using ehci-pci
[ 1522.728404] usb 1-1.5: New USB device found, idVendor=0a5c, idProduct=21e8, bcdDevice= 1.12
[ 1522.728408] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1522.728410] usb 1-1.5: Product: BCM20702A0
[ 1522.728412] usb 1-1.5: Manufacturer: Broadcom Corp
[ 1522.728414] usb 1-1.5: SerialNumber: 000272C5BBB4
[ 1522.837913] Bluetooth: hci0: BCM: chip id 63
[ 1522.838925] Bluetooth: hci0: BCM: features 0x07
[ 1522.854898] Bluetooth: hci0: BCM20702A
[ 1522.855900] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[ 1522.855919] bluetooth hci0: Direct firmware load for brcm/BCM20702A1-0a5c-21e8.hcd failed with error -2
[ 1522.855921] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0a5c-21e8.hcd not found
[ 1525.141916] Bluetooth: hci0: advertising data len corrected
[ 1525.145907] Bluetooth: hci0: advertising data len corrected
[ 1525.910055] input: M720 Triathlon as /devices/virtual/misc/uhid/0005:046D:B015.000A/input/input18
[ 1525.910220] hid-generic 0005:046D:B015.000A: input,hidraw0: BLUETOOTH HID v0.07 Keyboard [M720 Triathlon] on 00:02:72:C5:BB:B4
[ 1539.882814] usb 1-1.5: USB disconnect, device number 6
```
With
```
[ 1541.591423] usb 1-1.5: new full-speed USB device number 7 using ehci-pci
[ 1541.674285] usb 1-1.5: New USB device found, idVendor=0a5c, idProduct=21e8, bcdDevice= 1.12
[ 1541.674288] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1541.674291] usb 1-1.5: Product: BCM20702A0
[ 1541.674292] usb 1-1.5: Manufacturer: Broadcom Corp
[ 1541.674294] usb 1-1.5: SerialNumber: 000272C5BBB4
[ 1541.788794] Bluetooth: hci0: BCM: chip id 63
[ 1541.789784] Bluetooth: hci0: BCM: features 0x07
[ 1541.805801] Bluetooth: hci0: BCM20702A
[ 1541.806794] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[ 1542.795780] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1764
[ 1542.811696] Bluetooth: hci0: Broadcom Bluetooth Device
[ 1546.068327] input: M720 Triathlon as /devices/virtual/misc/uhid/0005:046D:B015.000B/input/input19
[ 1546.068535] hid-generic 0005:046D:B015.000B: input,hidraw0: BLUETOOTH HID v0.07 Keyboard [M720 Triathlon] on 00:02:72:C5:BB:B4
```

infirit, 2018-06-11:
Oh, forgot to add. There are shops that will sell guaranteed linux compatible dongles. For example https://www.thinkpenguin.com/gnu-linux/penguin-usb-bluetooth-40-micro-adapter-tpe-usbbluv4

lukehutch, 2018-06-11:
My log looks a bit more like your "With" log than your "Without" log -- at least I have no messages about errors loading firmware.

Thanks, I appreciate the link! I see that a number of adapters on Amazon now state specifically that they are supported under Linux. This shows Linux is finally relevant in the mainstream, I guess.