blueman-project/blueman #883: I am trying to connect my iphone6s to raspberrypi 3

rameshsabnis, 2018-06-11:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

infirit, 2018-06-18:
There is not enough in the report to help you. See https://github.com/blueman-project/blueman/wiki/Troubleshooting on what we need.

infirit, 2018-09-29:
Closing, we can reopen if details are provided.

closed 2018-09-29 by infirit