blueman-project/blueman #887: Use distutils / enttrypoints for scripts (apps)

blueyed, 2018-06-14:
I have noticed that `blueman-applet` and other scripts from `apps` use a shebang `#!/usr/bin/env python` (or `#!/usr/bin/env python3` on master).

This causes errors like the following when running it from some virtualenv:
```
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 12, in <module>
    import dbus
ModuleNotFoundError: No module named 'dbus'
```

By using `console_scripts` entry_points this would be avoided, since then a script would be generated to use the Python version where it was installed into, i.e. the system Python on my Arch Linux system.

On master the scripts in apps appear to be much smaller already, so maybe it's easy to use `console_scripts` from there.

This is a rather minor issue, but I've noticed it when running `blueman-applet` from my current terminal to see why it failed to start (bluetooth.service wasn't running, since I do not use it by default).

infirit, 2018-06-16:
Currently rebuilding my machine so quick email response.

You will have to satisfy all of blueman  dependencies in your virtualenv
environment. This includes binary Python modules like python-dbus and the
one from blueman. The shehang has nothing to do with this.

blueyed, 2018-06-16:
@infirit 
I've not installed blueman in the virtualenv - but it is installed system-wide.
But the shebang causes it to use the virtualenv's Python, which should not happen.

blueyed, 2018-06-17:
> If /usr/bin/env python results in blueman being run inside virtualenv your setup is broken.

No.
1. A virtualenv typically puts it `python` in front of `$PATH`.
2. `/usr/bin/env` looks up a program from the `$PATH`.

It is a rather minor issue, since typically it gets not run after a virtualenv is activated, but usually Python programs installed systemwide work then, too.

infirit, 2018-06-17:
>A virtualenv typically puts it python in front of $PATH.

I know but you said you are not running blueman in virtualenv so I fail to understand how virtualenv is causing problems for you systemwide. I am sure blueman is not the only python program using ``/bin/env python``.

blueyed, 2018-06-18:
I've said:

> This causes errors like the following when running it from some virtualenv:

i.e. a virtualenv is activated, and I run it from that shell.

> I am sure blueman is not the only python program using `/bin/env python`.

Yes, but that makes it not better really.
I have ~44 in /usr/bin, 8 starting with "blueman-", so you fix nearly 20% of them.. ;)

infirit, 2018-06-18:
>I have ~44 in /usr/bin, 8 starting with "blueman-", so you could fix nearly 20% of them.. ;)

See my answer on conflicting requests by distros. Raise it with the arch folks to rewrite the shebang.

>i.e. a virtualenv is activated, and I run it from that shell.

I don't see any point in running blueman from virtualenv like this. So you either build and run it in virtualenv with all deps satisfied or run it outside with the system python version.

I personally am not willing to support pick and mix and have users run into bugs/errors because the blueman module was build for a different version than it is run under.

blueyed, 2018-06-18:
Yes, Arch could patch the shebang in the scripts, but that would be required for every distro then basically.

When using `console_scripts` something like the following gets generated automatically (from flake8's system package):
```python
#!/usr/bin/python
# EASY-INSTALL-ENTRY-SCRIPT: 'flake8==3.5.0','console_scripts','flake8'
__requires__ = 'flake8==3.5.0'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('flake8==3.5.0', 'console_scripts', 'flake8')()
    )
```

I am not saying you should use `/usr/bin/python` yourself, but rather `console_script` entrypoints in `setup.py`.

>> i.e. a virtualenv is activated, and I run it from that shell.

> I don't see any point in running blueman from virtualenv like this.

This wasn't intentional, but since I am using Python a lot I often have a virtualenv activated, and then happened to try to see why `blueman-applet` wasn't starting.

> I personally am not willing to support pick and mix and have users run into bugs/errors because the blueman module was build for a different python version than it is run under.

Well, this is currently the case: `env` picks up `python` from `$PATH`, but that might not be the one that it was built for.
Why not use the proper way to install (wrapper) scripts using `console_scripts`?
What I am suggesting would exactly fix this.

cschramm, 2018-06-19:
Your request has nothing to do with `setuptools`. The shebangs in the generated `console_scripts` are just a side effect as it uses a shebang matching `sys.executable` by default. Regarding your flake8 system package, `setup.py build` was either run by `/usr/bin/python` or with `--executable=/usr/bin/python` or both.

So your actual request is to use the python path you're building blueman with for the shebang.

Using `/usr/bin/env` was primarily based on #187. Using the python environment blueman gets build with should fit that case as well. I actually cannot see a reason not to do that instead of using `/usr/bin/env` except for the necessary fuss to generate it.

cschramm, 2018-06-19:
One more thought: Using `console_scripts` to solve that is definitely not a good reason to use two build systems (the GNU build system currently in use or possibly meson (#844) and `setuptools`). I'd rather do the shebang generation with the build system we're using.

PR #1015 (2019-04-29): Set shebang to configured python interpreter
Closes #887
commit f9663be6 (2019-04-29): Set shebang to configured python interpreter (#1015)

Closes #887
closed 2019-04-29 by cschramm