blueman-project/blueman #888: no adapter found

KambhaktIshQ, 2018-06-17:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

> ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

KambhaktIshQ, 2018-06-17:
[image]

cschramm, 2018-06-17:
Check https://github.com/blueman-project/blueman/wiki/Troubleshooting#reporting-a-bug--asking-for-support and https://github.com/blueman-project/blueman/wiki/Troubleshooting#no-adapter-or-not-able-to-turn-on-bluetooth.

closed 2018-07-08 by infirit

Taariqq, 2018-07-16:
Same issue here. "no device connected" Even though when I plug the device into usb my bluetooth status goes to "on".

infirit, 2018-07-17:
@Taariqq, see the links @cschramm posted.

And I don't understand what you mean with ```Even though when I plug the device into usb my bluetooth status goes to "on".```.