blueman-project/blueman #891: Testing CI on gitlab

infirit, 2018-07-08:
My main frustration with github is its ancient CI image, xenial was release over two bloody years ago :sob:. So out of curiosity I decided to see how easy it was to do this on gitlab. It was surprisingly easy :open_mouth:.

1. import from github (issues with milestons, merge requests etc)
2. setup jobs for specific docker images
3. performance is pretty good

While it is possible to use docker in github it takes some effort, see https://docs.travis-ci.com/user/docker/.

See the result at https://gitlab.com/infirit/blueman and the yaml file at https://gitlab.com/infirit/blueman/blob/master/.gitlab-ci.yml.

cschramm, 2018-07-11:
I have no strong opinion on moving the project to GitLab, just want to point out that what you're criticizing is Travis CI and there are a lot of other options with GitHub integrations and free plans for FOSS projects. I'd expect the use of e.g. the popular CircleCI to have similar benefits.

infirit, 2018-07-11:
>I have no strong opinion on moving the project to GitLab

I was not suggesting it but good to know :+1: .

>you're criticizing is Travis CI 

I am yes, thanks for correcting me :smile:.

>the popular CircleCI to have similar benefits

Ooh, I was not aware of them. I'll try them out, thanks!

infirit, 2018-07-11:
See #893

closed 2018-07-11 by infirit