blueman-project/blueman #895: panel plugin dbus error

akostadinov, 2018-07-17:
blueman: blueman-2.1-0.9.alpha2.fc28.x86_64
BlueZ: bluez-5.50-1.fc28.x86_64
Distribution: Fedora 28
Desktop environment: XFCE 4.12
dbus-1.12.8-1.fc28.x86_64
dbus-glib-0.110-2.fc28.x86_64

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

```
blueman-applet 18.20.42 ERROR    PluginManager:209 Run       : Function on_adapt
er_property_changed on DiscvManager failed
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Base.py", line 102, in ge
t
    None)
GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.UnknownO
bject: Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Prope
rties" doesn't exist
 (41)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/blueman/main/PluginManager.py", line 20
6, in Run
    ret = getattr(inst, func)(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py", line 105, in on_adapter_property_changed
    self.update_menuitems()
  File "/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py", line 111, in update_menuitems
    elif (not self.adapter["Discoverable"] or self.adapter["DiscoverableTimeout"] > 0) and self.adapter["Powered"]:
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Base.py", line 137, in __getitem__
    return self.get(key)
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Base.py", line 108, in get
    raise parse_dbus_error(e)
blueman.bluez.errors.BluezDBusException: org.freedesktop.DBus.Error.UnknownObject Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

infirit, 2018-07-17:
I've seen this before when removing an adapter. I don't remember if I fixed but..

Can you confirm this actually happens when removing an adapter? If not when :smile:

akostadinov, 2018-07-17:
I never added or removed an adapter. This is a laptop that I suspend though so it might be the case.
Moreover I see this in the log that I missed the first time:
```
blueman-applet 18.20.42 ERROR    PluginManager:209 Run       : Function on_adapt
er_property_changed on DiscvManager failed
```

Maybe can close then.

akostadinov, 2018-07-17:
hmm, here is something on discovery:
```
blueman-applet 18.20.42 ERROR    PluginManager:209 Run       : Function on_adapt
er_property_changed on DiscvManager failed
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Base.py", line 102, in ge
t
    None)
GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.UnknownO
bject: Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Prope
rties" doesn't exist
 (41)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/blueman/main/PluginManager.py", line 206, in Run
    ret = getattr(inst, func)(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py", line 105, in on_adapter_property_changed
    self.update_menuitems()
  File "/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py", line 111, in update_menuitems
    elif (not self.adapter["Discoverable"] or self.adapter["DiscoverableTimeout"] > 0) and self.adapter["Powered"]:
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Base.py", line 137, in __getitem__
    return self.get(key)
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Base.py", line 108, in get
    raise parse_dbus_error(e)
blueman.bluez.errors.BluezDBusException: org.freedesktop.DBus.Error.UnknownObject Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

This is on remove:
```
blueman-applet 18.20.42 ERROR    PluginManager:209 Run       : Function on_adapter_removed on DiscvManager failed
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/blueman/main/PluginManager.py", line 206, in Run
    ret = getattr(inst, func)(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py", line 77, in on_adapter_removed
    self.init_adapter()
  File "/usr/lib/python3.6/site-packages/blueman/plugins/applet/DiscvManager.py", line 67, in init_adapter
    self.adapter = self.parent.Manager.get_adapter()
  File "/usr/lib/python3.6/site-packages/blueman/bluez/Manager.py", line 85, in get_adapter
    raise DBusNoSuchAdapterError("No adapter(s) found")
blueman.bluez.errors.DBusNoSuchAdapterError: No adapter(s) found
```
Seems like on remove the `ss` error is not threre.

Maybe I should have used pastebin. But then we lose ability of search engines to find the bug. Doesn't github provide expandable code blocks to avoid page clutter while retaining logs where they belong?

infirit, 2019-05-04:
This should be fixed now and we are about to release the first beta. Closing but if you run into this in the beta please report back.

closed 2019-05-04 by infirit

kuraga, 2023-01-07:
Similar: https://github.com/blueman-project/blueman/issues/1938#issuecomment-1356125774