blueman-project/blueman #902: 2.1.alpha3

cschramm, 2018-08-02:
We currently have 89 commits in master since 2.1.alpha2 (https://github.com/blueman-project/blueman/compare/2.1.alpha2...master) including quite some bug fixes and improvements, see e.g. https://github.com/blueman-project/blueman/issues/895#issuecomment-405762518.

We should probably release another version. Unfortunately I still consider 2.1 in alpha state as we're still in the middle of the GDBus migration. So it'll be 2.1.alpha3.

What needs to be done is adding something useful to the the changelog as we did not maintain it in master.

Anything else?

infirit, 2018-08-02:
Yes, good idea. I introduced some nasty bugs in alpha2 :sob: 

>GDBus migration

The service works for methods and signals. The only thing not working is properties which we do not use. I got really annoyed with it but have been poking at it once in a while to see if I could do my own implementation that integrates better with plugins. I have not found anything that I like.

>Anything else?

Nothing really, perhaps the nmdun bits which I can rebase and see if people run into problems opening issues here :smile:

infirit, 2018-08-09:
I don't know when I have PPPSupport figured out. Do a new alpha now or you want to wait?

infirit, 2018-08-19:
Another one, #910. When merged I'll update the changelog for the release.

cschramm, 2018-12-24:
I think we're in a good state for a pre-release. Anything else we should take care of in addition to the changelog?

infirit, 2018-12-24:
I like to bump the NetConf class version. We made backwards incompatible changes which I ran into testing 2.0.7 (saw the release :+1: ). I'll open a PR with this change and an update to CHANGELOG.

closed 2019-03-30 by cschramm