blueman-project/blueman #904: Disable BT on startup

rypz79, 2018-08-05:
blueman: 2.0.5
BlueZ: 5.48
Distribution: Ubuntu 18.04 
Desktop environment: GNOME 3.28.2
This tutorial from [https://askubuntu.com/questions/67758/how-can-i-deactivate-bluetooth-on-system-startup](https://askubuntu.com/questions/67758/how-can-i-deactivate-bluetooth-on-system-startup)

**1. block bluetooth in rc.local**
Add this line at the end of the file `/etc/rc.local` right before the line `exit 0`:
    rfkill block bluetooth
This will do the trick, but only in older Ubuntu versions using **upstart**.

**2. reenable rc.local on systemd**
On Ubuntu 15.10 with **systemd** as startup manager the `/etc/rc.local` file is not used by default any more, so call this on a terminal:
`sudo systemctl edit --full rc-local`
Which opens an editor. At the end of the file i added
```
 [Install]
 WantedBy=multi-user.target
```
Those changes are activated by
`sudo systemctl reenable rc-local`
Now the file `/etc/rc.local` is used as you know it.

**3. disable blueman-applet on start**
```
sudo sed -i 's/NoDisplay=true/NoDisplay=false/g' /etc/xdg/autostart/blueman.desktop
gnome-session-properties
```
Now blueman-applet is visible and you can disable it from startup programs
**4. Reboot to see the effect**

Worked, but it involves disabling the applet which becomes candidate to be autoremoved when upgrading apt's. How can I keep the applet working while making BT default state OFF at startup ?

Thank you
* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2018-08-05:
blueman's `PowerManager` plugin has a setting for powering up adapters when starting which is currently active by default. We're eager to improve that as it's a constant source for confusion. Feel free to contribute to #877.

When the blueman setting is turned off, you can disable bluetooth with `rfkill block bluetooth` or using the blueman applet which does the same thing. The rfkill state should survive reboots. No need to add it to some init script. Works fine for me on Debian.

rypz79, 2018-09-07:
Solved:
`sudo vim /etc/rc.local`
insert these commands at the end of the file and save
```
bluetooth off
restart 0
```
make sure the file is executable `sudo chmod +x /etc/rc.local`
logout or reboot

infirit, 2018-09-28:
Closing, if folks want to contribute to the discussion please do so in #877

closed 2018-09-28 by infirit